riscv-collab/riscv-gnu-toolchain #930: Error: illegal operands `sw ra,0(base)'

guoxiangCN, 2021-07-24:
i wrote a macro in a asm file, but the Assembler notice me the error msg:
Error: illegal operands `sw ra,0(base)'
My codes:
`.macro reg_save base
	sw ra, 0(base)
	sw sp, 4(base)
	sw gp, 8(base)
	sw tp, 12(base)
	sw t0, 16(base)
	sw t1, 20(base)
	sw t2, 24(base)
	sw s0, 28(base)
	sw s1, 32(base)
	sw a0, 36(base)
	sw a1, 40(base)
	sw a2, 44(base)
	sw a3, 48(base)
	sw a4, 52(base)
	sw a5, 56(base)
	sw a6, 60(base)
	sw a7, 64(base)
	sw s2, 68(base)
	sw s3, 72(base)
	sw s4, 76(base)
	sw s5, 80(base)
	sw s6, 84(base)
	sw s7, 88(base)
	sw s8, 92(base)
	sw s9, 96(base)
	sw s10, 100(base)
	sw s11, 104(base)
	sw t3, 108(base)
	sw t4, 112(base)
	sw t5, 116(base)
	sw t6, 120(base)
.endm
`

jim-wilson, 2021-07-24:
See the assembler docs on how to write a macro, or look at examples in for instance newlib.
https://sourceware.org/binutils/docs-2.37/as/Macro.html#Macro
https://sourceware.org/git/?p=newlib-cygwin.git;a=blob;f=libgloss/rx/rxsys.h;h=39c5373eced73997f4c05891c668124890673832;hb=HEAD#l41
Your macro is missing backslashes for argument substitution.

guoxiangCN, 2021-07-25:
> See the assembler docs on how to write a macro, or look at examples in for instance newlib.
> https://sourceware.org/binutils/docs-2.37/as/Macro.html#Macro [...]

Thanks!!!

closed 2021-07-30 by guoxiangCN