riscv-collab/riscv-gnu-toolchain #931: libc-start.c/sysdep.h Error in building rv32e toolchain

gsh1983, 2021-07-25:
I am trying to build the following configuration on Ubuntu 20.04.2.0 LTS:
./configure --prefix=/opt/riscv --with-arch=rv32emac --with-abi=ilp32e
make linux

Getting below Error:
libc-start.c: In function '__libc_start_main':
../sysdeps/unix/sysv/linux/riscv/sysdep.h:190:27: error: register specified for '__a7' isn't suitable for data type
  190 |         register long int __a7 asm ("a7") = number;                     \

Can someone please help how to resolve this?

kito-cheng, 2021-07-25:
glibc and linux didn't support rv32e, so...the issue is hard to resolve until someone spend lots of effort to port glibc and linux kernel to rv32e.

jim-wilson, 2021-07-25:
use "make" instead of "make linux" to build an embedded elf toolchain

TommyMurphyTM1234, 2022-09-27:
Incorrect build configuration - Linux toolchain does not support rv32e arch/abi.

closed 2022-09-27 by TommyMurphyTM1234