riscv-collab/riscv-gnu-toolchain #932: How to build riscv-gnu toolchain for windows?

panciyan, 2021-07-26:
Hi,
Currently, I'm trying to build a riscv-gnu toolchain with flag "-enable-multilib" on Windows 10 msys2 enviroment, and the toolchain needs to include both *-unknown-elf-* (baremetal) and *-unknown-linux-* (linux) parts, it is blocked at glibc compile stage. Does anyone can help the issue? Or show a guide how to build riscv-gnu linux part on Windows 10? Thanks o lot!

Detail glibc compile error report:
if test -r /c/IDE-103/src/build-glibc-linux-rv32imac-ilp32/csu/abi-tag.h.new; then mv -f /c/IDE-103/src/build-glibc-linux-rv32imac-ilp32/csu/abi-tag.h.new /c/IDE-103/src/build-glibc-linux-rv32imac-ilp32/csu/abi-tag.h; \
else echo >&2 'This configuration not matched in ../abi-tags'; exit 1; fi
python3 -B ../scripts/gen-as-const.py --cc="riscv64-unknown-linux-gnu-gcc -march=rv32imac -mabi=ilp32 -std=gnu11 -fgnu89-inline  -mcmodel=medlow -g -O2 -march=rv32imac -mabi=ilp32 -Wall -Wwrite-strings -Wundef -fmerge-all-constants -frounding-math -fno-stack-protector -Wstrict-prototypes -Wold-style-definition -fmath-errno     -ftls-model=initial-exec      -I../include -I/c/IDE-103/src/build-glibc-linux-rv32imac-ilp32/csu  -I/c/IDE-103/src/build-glibc-linux-rv32imac-ilp32  -I../sysdeps/unix/sysv/linux/riscv/rv32/  -I../sysdeps/unix/sysv/linux/riscv  -I../sysdeps/riscv/nptl  -I../sysdeps/unix/sysv/linux/generic/wordsize-32  -I../sysdeps/unix/sysv/linux/generic  -I../sysdeps/unix/sysv/linux/riscv/rv32  -I../sysdeps/unix/sysv/linux/include -I../sysdeps/unix/sysv/linux  -I../sysdeps/nptl  -I../sysdeps/pthread  -I../sysdeps/gnu  -I../sysdeps/unix/inet  -I../sysdeps/unix/sysv  -I../sysdeps/unix  -I../sysdeps/posix  -I../sysdeps/riscv/rv32/  -I../sysdeps/riscv/rv32  -I../sysdeps/riscv/nofpu  -I../sysdeps/ieee754/soft-fp  -I../sysdeps/riscv  -I../sysdeps/init_array  -I../sysdeps/ieee754/ldbl-128  -I../sysdeps/ieee754/dbl-64  -I../sysdeps/ieee754/flt-32  -I../sysdeps/wordsize-32  -I../sysdeps/ieee754  -I../sysdeps/generic  -I.. -I../libio -I. -nostdinc -isystem c:/ide-103/src/out_0723/bin/../lib/gcc/riscv64-unknown-linux-gnu/10.1.0/include -isystem c:/ide-103/src/out_0723/bin/../lib/gcc/riscv64-unknown-linux-gnu/10.1.0/include-fixed -isystem /c/IDE-103/src/linux-headers/include -D_LIBC_REENTRANT -include /c/IDE-103/src/build-glibc-linux-rv32imac-ilp32/libc-modules.h -DMODULE_NAME=libc -include ../include/libc-symbols.h       -DTOP_NAMESPACE=glibc -DGEN_AS_CONST_HEADERS \
                   -MD -MP -MF /c/IDE-103/src/build-glibc-linux-rv32imac-ilp32/tcb-offsets.h.dT \
                   -MT '/c/IDE-103/src/build-glibc-linux-rv32imac-ilp32/tcb-offsets.h.d /c/IDE-103/src/build-glibc-linux-rv32imac-ilp32/tcb-offsets.h'" \
          **../sysdeps/riscv/nptl/tcb-offsets.sym > /c/IDE-103/src/build-glibc-linux-rv32imac-ilp32/tcb-offsets.hT
riscv64-unknown-linux-gnu-gcc: error: /c/IDE-103/src/build-glibc-linux-rv32imac-ilp32/tcb-offsets.h': No such file or directory**

jim-wilson, 2021-07-26:
The elf toolchain can be built on Windows.  The linux toolchain can't.  At least not easily.  The linux toolchain requires a case-sensitive file system, and WIndows file systems are case-insensitive by default.  You might be able to fix this by changing filesystem settings, but then this is likely to break Windows tools.  You might also be able to fix this by hacking the sources so that they don't depend on case, but this is complicated, and may have to be redone everytime we upgrade an affected part of the toolchain.

It is easier to build a windows hosted riscv linux targetted toolchain on a linux build machine.  This is called a canadian cross build.

In the glibc makefiles, it uses both .os and .oS as file name extensions, and these must be distinct files or the build will fail.  There may also be other similar cases.

panciyan, 2021-07-26:
Thanks Wilson, like you said, elf toolchain have been built out on my Windows OS. For canadian cross build linux toolchain, do you have any detail instructions? Do I need to rewite the makefile?

jim-wilson, 2021-07-26:
I don't think we have any support for generic canadian cross build linux toolchain from riscv-gnu-toolchain.  The general process is that you build a x86-linux cross windows compiler, and a x86-linux cross riscv compiler, and then use both of those to build a windows cross riscv compiler.

crosstools-NG has some support for building canadian cross compilers.
[https://crosstool-ng.github.io/docs/toolchain-types/](url)

riscv-gnu-toolchain is only useful for building toy compilers that generate code that can be run on a simulator.  If you have an actual linux target that you want to compile code for, you should not use riscv-gnu-toolchain.  In this case, you should instead do a sysroot style build.  A sysroot is a collection of files from the target root filesystem that contains everything the compiler needs, generally /usr/include, /lib, and /usr/lib, and /lib64 and /usr/lib64 if they exist.  You then only need to build FSF binutils and FSF gcc, using the --sysroot=<path> configure option for both binutils and gcc.  Note that in this case you don't need a canadian cross build as you aren't building glibc.  So this could be done directly on a windows machine and hence is likely easier.  But you need the sysroot to start, so you need a riscv64-linux system to extract the sysroot from.  Or if you only want to use a simulator, you could use riscv-gnu-toolchain built on linux to create the sysroot.

panciyan, 2021-07-27:
Hi Jim,
I'm confused. In riscv-gnu toochain build process:
1.  ./configure xxxxxx
2.  make                        ----> build newlib related toochain
3.  make linux                ----> build linux related toolchain
In step 3 "make linux", I notice that in gnu project, there're already linux-headers files. Per my understanding, the headers and glibc build would install to sysroot, so I can use the gnu project directly to "make linux" in msys2 compile ENV even I don't have real riscv target. Is it right?

jim-wilson, 2021-07-27:
You have to build glibc to create a sysroot, and glibc requires a case-sensitive file system in order to build it.   Most of /usr/include and some important bits of /lib and /usr/lib come from glibc.

wusi524888, 2023-10-16:
can you solved it ?

TommyMurphyTM1234, 2023-10-16:
> can you solved it ?

Solve what specifically?