riscv-collab/riscv-gnu-toolchain #933: Is reusing a variable good practice for memory usage in your compiler?

Kian75, 2021-08-03:
Hi... I have written a code in C for Convolutional Neural Networks (CNN) and after each layer some variables become unused. So do I gain memory if I reuse them in next layers instead of defining new variables or I gain nothing in terms of memory at all?

jim-wilson, 2021-08-05:
A good optimizing compiler will do live-range analysis to determine when variables are in use, and register allocation will take that into account.  So in general, it shouldn't make any difference whether you reuse variables or not.

One possible exception is if you have local variables and take the address of them.  Tracking live ranges for stack slots is much harder.  In this case, you might benefit from reusing a variable.  But adding brace levels to the code is just as easy and will usually work as well.  E.g. if you have " ... { int i; ...&i ... } { int j; ...&j ... }" then the compiler can assume that i and j can share space even if if it can't prove that &i dies before the first use of j.  This does require that you write good code that handles local variable pointers correctly though.

In general, I would suggest writing code that is readable and maintainable first.  And then only make it ugly it the compiler isn't optimizing as well as you expected.

TommyMurphyTM1234, 2022-09-27:
Coding question - not an issue with the RISC-V GCC toolchain.

closed 2022-09-27 by TommyMurphyTM1234