riscv-collab/riscv-gnu-toolchain #935: Corrupted DWARF expression while debugging gdb

atishp04, 2021-08-05:
I keep getting this error while inspecting any value in gdb while debugging latest kernel
gdb.error: dwarf2_find_location_expression: Corrupted DWARF expression.

Any idea what's wrong ? I tried rebuilding from the latest sources of both this repo and upstream gdb 10.1.
Both resulted in the same error.

jim-wilson, 2021-08-05:
New versions of gcc are emitting dwarf that old versions of gcc never emitted before, and which gdb thus did not have support for.  Until it was recently added to gdb.  GCC was "broken" by this change.
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=99090
gdb was fixed by this change
https://sourceware.org/pipermail/gdb-patches/2021-June/180193.html
You need gcc 11 or maybe gcc 12 to see the problem.

You can also find the patch in freedom-u-sdk as I asked for it to be added there.
https://github.com/sifive/meta-sifive/blob/2021.07/recipes-devtools/gdb/files/0001-gdb-Support-DW_LLE_start_end.patch

If you hit some other problem, then I need info on how to reproduce.

atishp04, 2021-08-09:
Thanks. Closing the issue.

closed 2021-08-09 by atishp04