riscv-collab/riscv-gnu-toolchain #938: [question] Intrinsics of RISC-V extension P

gquintin, 2021-08-10:
Hi RISC-V,

I do not know if it is the right place to ask my question. I am the maintinaer of NSIMD (<https://github.com/agenium-scale/nsimd>) and we are including RSIC-V extension V intrinsics into the library. I am not aware of intrinsics for the P extension. Are there any? What are your plans about them?

jim-wilson, 2021-08-10:
I would suggest asking in the P group.  You can file a github issue to ask a question.
github.com/riscv/riscv-p-spec

I do see intrinsics defined in the spec for each instruction.  Though the intrinsic names don't include a p prefix which may have to change for more consistent intrinsic naming across various extensions.

There are pull requests for p support for riscv-binutils and risccv-gcc where the work is being done.  See for instance
https://github.com/riscv/riscv-gcc/pull/258

TommyMurphyTM1234, 2022-09-29:
@kito-cheng - I presume that there's no specific plan to integrate this PR imminently?

* https://github.com/riscv-collab/riscv-gcc/pull/25

Would it make any sense to open an issue here and close this one?

* https://github.com/riscv/riscv-p-spec/issues

cmuellner, 2022-09-29:
The P extension is neither ratified nor frozen.
So all enablement patchsets are experimental (as the spec can - and most likely will - change).
For Binutils/GCC that means that the patchsets won't get merged as currently only patches for at least frozen standard extensions are accepted (to meet the stability requirements of the upstream projects).

Therefore I would recommend `nsimd` to ignore the P extension at the moment.
Adding support for the P extension can be considered once the P extension(s) will be frozen.

closed 2022-09-29 by cmuellner

TommyMurphyTM1234, 2022-09-29:
Thanks @cmuellner.