riscv-collab/riscv-gnu-toolchain #939: Compile not working

nikita890-art, 2021-08-11:
Hi all!

I have built the riscv gnu toolchain on my ubuntu 

I have made some changes in some of the codes in the toolchain as i am trying to add mod instruction in the toolchain.
I am following the procedure as shown in this website,

https://nitish2112.github.io/post/adding-instruction-riscv/

Now after making the changes, I have to compile the toolchain again,

So I type,
sudo make

but I get the message,
nothing to be done for "all"

I want to compile the updated toolchain.
How do I do that?
Please help!

jim-wilson, 2021-08-11:
The riscv-gnu-toolchain Makefile uses stamp files in the stamp dir to avoid recompiling things that were already built.  You can delete the appropriate stamp file and rebuild something from scratch.  This may cause other dependent dirs to be rebuilt from scratch.  E.g. if you force a binutils rebuild, that will also force gcc and C library rebuilds as they depend on binutils.

The top level makefile doesn't support incremental builds, but if you want to do this by hand, you can cd into the dir you want to incrementallyl rebuild, and then run make and "make install" in there.  There is one detail here which is in that in the stage1 build dir we only do "make all-gcc" and "make install-gcc".  If you later type make in the top level dir, then the makefile will notice that the hand built dir changed, and then rebuild from scratch all dependent dirs.

You can also just do "make clean" followed by "make" to rebuild everything from scratch.

I would suggest that you configure using a --prefix that you have write access to so that you don't need to use sudo for the build.  Personally, I like to use --prefix=`pwd`/build-install to put the install dir inside the build tree.  This is useful if you have multiple build trees.

nikita890-art, 2021-08-11:
Ok, sir @jim-wilson 
I have to update the gnu toolchain libraries often and I think reinstalling the entire toolchain over and over again each time I update something is time-consuming. Is there any way that I can modify the toolchain in a faster way? And if so, how do I do it?
Thank you

jim-wilson, 2021-08-11:
You can incrementally recompile the part you are modifying as mentioned, e.g. binutils and gcc.  But there is no easy way to handle the C library.  It needs to be recompiled from scratch each time if you want it to use the new instruction.  Or you can just choose not to recompile it if it doesn't need to use the new instruction.  Just don't run make at the toplevel after the original build, and only run make and make install in the subdirs you want to rebuild.

nikita890-art, 2021-08-11:
I will be updating the toolchain to add new instructions. So you mean there is no other way that recompiling the entire toolchain over again?

nikita890-art, 2021-08-11:
And if it is not too much trouble sir, can u state the commands that I need to use to recompile the gnu toolchain after I have modified the files to add a new instruction?
@jim-wilson

jim-wilson, 2021-08-11:
I would suggest getting your changes working first before trying to rebuild the C library.  This is a waste of time if your changes aren't working yet.

cd build-binutils-newlib
make 
make install
cd ..
cd build-gcc-newlib-stage1
make all-gcc
make install-gcc
cd ..

Then test whether the new instruction support works.  When it does work, try rebuilding the whole thing from scratch using make clean and then make.

nikita890-art, 2021-08-11:
Ok, sir. Thank you.
Will the make work even if there is a syntax or semantic error in the updated code or will it throw an error while compiling?

You said to check if the new instruction support works, by that do u mean to run a c code and check the dump if the instruction is used?

And also, just to clarify, the make clean and the make commands need to be run inside the build folder present in the toolchain ryt?

Thank you

@jim-wilson

jim-wilson, 2021-08-11:
If there is a syntax error, the make will fail with an error, regardless of which method you are using to build.

Yes, use a C testcase and objdump to verify the change works.

make clean at the top level will remove everything except the configuration info.  Make clean in a subdir may or may not work, but will only remove stuff in that subdir.

nikita890-art, 2021-08-11:
But if the changes work, then why is there a need to make clean and make again?
@jim-wilson

nikita890-art, 2021-08-14:
Thank you sir @jim-wilson 
It works fine now!

closed 2021-08-14 by nikita890-art