riscv-collab/riscv-gnu-toolchain #940: Disable compression of `jalr` instructions into `c.jalr` instructions when the c extension is enabled

IdanHo, 2021-08-11:
Hey there! I'm trying to compile an assembly file using the riscv-gnu-toolchain's gcc with c extension support enabled, but im facing the following issue: Even thought i've specificed `-O0` to disable all optimizations it seems like gcc's assembler is replacing all instances of the non-compressed `jalr` instruction with the compressed `c.jalr` variant for example: `jalr x25` gets compiled into 0x9c82 (which is `c.jalr x25`), is there some kind of option i can enable to disable the automatic compression of instructions but to still allow the usage of compressed instruction explicitly (i.e. i should still be able to write `c.jalr x25` myself if i wanted to), the full gcc command im using is as follows:
```shell
/usr/local/bin/riscv64-unknown-elf-gcc -g -march=rv64gc -mabi=lp64d -Wall -Wundef -Werror -ffreestanding -mcmodel=medany -Wfatal-errors -O0 -o test.S.obj -c test.S
```

jim-wilson, 2021-08-11:
".option norvc" turns off assembler compression.  ".option rvc" turns it on.  It is usually better to use push/pop so you don't need to know the current state.  So you can do
  .option push
  .option norvc
  ...
  .option pop
and return to the original state.

IdanHo, 2021-08-11:
Nice, this worked great, thanks a bunch! :)

closed 2021-08-11 by IdanHo