riscv-collab/riscv-gnu-toolchain #941: What is the correct way to generate 4 different architectures, rv32i, rv32im, rv32imc, rv32imcv, all ilp32?

carlosdelfino, 2021-08-12:
Dear colleagues.

I'm trying to generate my own toolchain on an Ubuntu 18.04 that runs on a Dual Core, where there is no other Risc-V library or tool installed or compiled, and I need to generate 4 distinct architectures that can be used separately:

* rv32i,
* rv32im,
* rv32imc,
* rv32imcv,

All with ABI **ilp32**.

I tried the following parameterization formats, following the examples suggested in the README.md file:

* `–with-multilib-generator=”rv32ic-ilp32--m*v”`, this compiled but when I used it it just worked --march=rv32ic
* ` –with-multilib-generator=”rv32i-ilp32--mc*mcv”`
* `--with-multilib-generator=rv32i-ilp32--;rv32imv-ilp32--;rv32imc-ilp32--`

What would be the best way to use such parameter so that I generate 4 indistinct groups of architectures, so that when they are used in assembly code generation I can compare their results one by one?

carlosdelfino, 2021-08-12:
I remove "c" extension, it a mistake use it now.

carlosdelfino, 2021-08-12:
@kito-cheng tanks.

Now think I understand, I need to generate some libraries first to use them in the generation of the others.

I thought I could generate everything at the same time.

Should I also generate the executable from gcc choose? it seems to me that this is implied in the text and I didn't realize it.

carlosdelfino, 2021-08-12:
> --with-multilib-generator="rv32i-ilp32--mc*mcv": Build one multi-lib for rv32ic/ilp32 and reuse that when you compile with rv32imc/ilp32 and rv32imcv/ilp32
> 
> --with-multilib-generator="rv32i-ilp32--;rv32imv-ilp32--;rv32imc-ilp32--": Build 3 multi-lib for rv32ic/ilp32, rv32imv/ilp32 and rv32imc/ilp32.
> 
> So in your case I would suggest you using --with-multilib-generator="rv32i-ilp32--;rv32imv-ilp32--;rv32imc-ilp32--;rv32imcv-ilp32--".

in the suggestions above, it is `--with-multilib-generator="rv32i-ilp32--....`

and not: `--with-multilib-generator="rv32ic-ilp32--....`?

For me to do these test builds it will take one to two weeks, so I ask before testing.

jim-wilson, 2021-08-12:
You can use "riscv64-unknown-elf-gcc --print-multi-lib" to see what multilibs are supported.  You don't need to build the entire toolchain, you can just build "make stamps/build-gcc-newlib-stage1".  You can also check build-gcc-newlib-stage1/gcc/multilib.h which has the library reuse info.

carlosdelfino, 2021-08-12:
Before I posted this issue, I had started a compilation with the `` parameters, and it ended now and gave the following error:

```
/home/carlosdelfino/workspace/RISCuinho/riscuinho-gnu-toolchain/rv32i-rv32im-ilp32/build-gcc-linux-stage2/./gcc/xgcc -B/home/carlosdelfino/workspace/RISCuinho/riscuinho-gnu-toolchain/rv32i-rv32im-ilp32/build-gcc-linux-stage2/./gcc/ -B/opt/riscuinho/riscv64-unknown-linux-gnu/bin/ -B/opt/riscuinho/riscv64-unknown-linux-gnu/lib/ -isystem /opt/riscuinho/riscv64-unknown-linux-gnu/include -isystem /opt/riscuinho/riscv64-unknown-linux-gnu/sys-include    -O2 -mcmodel=medlow -march=rv32imac -mabi=ilp32 -O2  -O2   -mcmodel=medlow -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition  -isystem ./include   -fPIC -g -DIN_LIBGCC2 -fbuilding-libgcc -fno-stack-protector   -fPIC -I. -I. -I../../../.././gcc -I../../../../../../../riscv-gnu-toolchain/riscv-gcc/libgcc -I../../../../../../../riscv-gnu-toolchain/riscv-gcc/libgcc/. -I../../../../../../../riscv-gnu-toolchain/riscv-gcc/libgcc/../gcc -I../../../../../../../riscv-gnu-toolchain/riscv-gcc/libgcc/../include  -DHAVE_CC_TLS  -o _negdi2.o -MT _negdi2.o -MD -MP -MF _negdi2.dep -DL_negdi2 -c ../../../../../../../riscv-gnu-toolchain/riscv-gcc/libgcc/libgcc2.c -fvisibility=hidden -DHIDE_EXPORTS
In file included from /opt/riscuinho/sysroot/usr/include/features.h:497,
                 from /opt/riscuinho/sysroot/usr/include/bits/libc-header-start.h:33,
                 from /opt/riscuinho/sysroot/usr/include/stdio.h:27,
                 from ../../../../../../../riscv-gnu-toolchain/riscv-gcc/libgcc/../gcc/tsystem.h:87,
                 from ../../../../../../../riscv-gnu-toolchain/riscv-gcc/libgcc/libgcc2.c:27:
/opt/riscuinho/sysroot/usr/include/gnu/stubs.h:8:11: fatal error: gnu/stubs-ilp32.h: No such file or directory
    8 | # include <gnu/stubs-ilp32.h>
      |           ^~~~~~~~~~~~~~~~~~~
compilation terminated.
Makefile:501: recipe for target '_negdi2.o' failed
make[5]: *** [_negdi2.o] Error 1
make[5]: Leaving directory '/home/carlosdelfino/workspace/RISCuinho/riscuinho-gnu-toolchain/rv32i-rv32im-ilp32/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/lib32/ilp32/libgcc'
Makefile:1211: recipe for target 'multi-do' failed
make[4]: *** [multi-do] Error 1
make[4]: Leaving directory '/home/carlosdelfino/workspace/RISCuinho/riscuinho-gnu-toolchain/rv32i-rv32im-ilp32/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libgcc'
Makefile:125: recipe for target 'all-multi' failed
make[3]: *** [all-multi] Error 2
make[3]: Leaving directory '/home/carlosdelfino/workspace/RISCuinho/riscuinho-gnu-toolchain/rv32i-rv32im-ilp32/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libgcc'
Makefile:12855: recipe for target 'all-target-libgcc' failed
make[2]: *** [all-target-libgcc] Error 2
make[2]: Leaving directory '/home/carlosdelfino/workspace/RISCuinho/riscuinho-gnu-toolchain/rv32i-rv32im-ilp32/build-gcc-linux-stage2'
Makefile:958: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/carlosdelfino/workspace/RISCuinho/riscuinho-gnu-toolchain/rv32i-rv32im-rv32im-ilp32/build-gcc-linux-stage2'
Makefile:389: recipe for target 'stamps/build-gcc-linux-stage2' failed
make: *** [stamps/build-gcc-linux-stage2] Error 2
```

I will try again now with the correct parameterization suggested.

jim-wilson, 2021-08-12:
You didn't mention that this was a linux compiler.  I assumed you were doing embedded elf builds.  Multilib for glibc is much more difficult than multilib for newlib.  The issue with gnu/stubs-ilp32.h is something you may need to fix by hand after the build if you do the build this way.  Linux distros handle this in their packaging system.  riscv-gnu-toolchain doesn't have a solution for it.

Anyways, to fix this issues with the stubs fils, you need to create a /usr/include/gnu/stubs.h file that includes the appropriate stubs-*.h file depending on predefined ABI macros, and you need to ensure that all of the stubs-*.h files are in there.  You might need to do multiple glibc builds in order to get all of the stubs-*.h files that you need.

Since you are building only 32-bit, adding --with-arch=rv32i --with-abi=ilp32 configure options might help to get the default glibc build compatible with the multilibs.  Since you are only using one ABI, you only need one set of stubs*.h files so this might work our OK.

But a better solution may be to just build 4 different linux compilers if you need 4 different arch/abi combinations.

If you want to try to fix this, the first step is to look at the include/gnu dir in the installed sysroot to see what you have, and what the potential problems there might be.

carlosdelfino, 2021-08-13:
So after I generate the gcc executable/binary, the libraries can be added by changing the `--with-multilib-generator` parameter?

jim-wilson, 2021-08-13:
You must choose the multilib set before building gcc, as gcc needs to know what the choices are, and how to map arch/abi options to those choices.

carlosdelfino, 2021-08-13:
Now I have had success with several possible architecture/abi combinations. But I can't add new combinations once I've built a set, even removing the `stamps/build-gcc-newlib-stage1` directory and running again `make stamps/build-gcc-newlib-stage1`, you always need to rebuild from of `make newlib`, is the process anyway, or is there something wrong I'm doing?.

But I still haven't quite assimilated this process of identifying the extensions, where you use the identifiers after `--` ie what the practical effect is when using `rv32i--m*c` or other similar variations.

jim-wilson, 2021-08-13:
It isn't clear what your problem is.  Note that rebuilding gcc stage1 only rebuilds the compiler.  It doesn't rebuild the gcc libraries (like libgcc) or the C library.  So if you are trying to link an execuble, and need new multilib libgcc and C libraries, then you need to rebuild both gcc and the C library.  That requires rebuilding gcc stage1, newlib, and gcc stage2.  "make newlib" is the easiest way to do that.

If you just want to experiment with --with-multilib-generator, then you really only need stage1 gcc, and then look at multilib.h file, and gcc --print-multi-lib, and use gcc -v -march=... -mabi=... and look at the linker command to see if you are getting the result you want.  That assumes you have some understanding of how gcc works.  If you are still figuring things out, it might be easier to just rebuild everything.

carlosdelfino, 2021-08-13:
I'm going to close this issue, as the main theme that led me to open it has largely disappeared.

About the generation of libraries as the parameter is changed, I will do more tests.

Thank you for your help.

closed 2021-08-13 by carlosdelfino