riscv-collab/riscv-gnu-toolchain #942: Compile branch(rvv-intrinsic)

vigoss30611, 2021-08-12:
I run command like this:
./configure --prefix=/home/sun/output/ --with-arch=rv64imafdv --with-abi=lp64
make newlib
Anyone who know how to solve this problem? error message like this:
make[5]: Leaving directory '/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib'
Making all in libc
make[5]: Entering directory '/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/libc'
Making all in argz
make[6]: Entering directory '/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/libc/argz'
riscv64-unknown-elf-gcc -B/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/ -isystem /home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/targ-include -isystem /home/sun/riscv-gnu-toolchain/riscv-newlib/newlib/libc/include -B/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/libgloss/riscv64 -L/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/libgloss/libnosys -L/home/sun/riscv-gnu-toolchain/riscv-newlib/libgloss/riscv64    -DPACKAGE_NAME=\"newlib\" -DPACKAGE_TARNAME=\"newlib\" -DPACKAGE_VERSION=\"3.2.0\" -DPACKAGE_STRING=\"newlib\ 3.2.0\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -I. -I/home/sun/riscv-gnu-toolchain/riscv-newlib/newlib/libc/argz -fno-builtin -DHAVE_NANOSLEEP      -O2 -D_POSIX_MODE   -mcmodel=medlow -c -o lib_a-dummy.o `test -f 'dummy.c' || echo '/home/sun/riscv-gnu-toolchain/riscv-newlib/newlib/libc/argz/'`dummy.c
Assembler messages:
Fatal error: invalid -march= option: `rv64imafdv'
make[6]: *** [Makefile:393: lib_a-dummy.o] Error 1
make[6]: Leaving directory '/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/libc/argz'
Making all in stdlib
make[6]: Entering directory '/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/libc/stdlib'
riscv64-unknown-elf-gcc -B/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/ -isystem /home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/targ-include -isystem /home/sun/riscv-gnu-toolchain/riscv-newlib/newlib/libc/include -B/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/libgloss/riscv64 -L/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/libgloss/libnosys -L/home/sun/riscv-gnu-toolchain/riscv-newlib/libgloss/riscv64    -DPACKAGE_NAME=\"newlib\" -DPACKAGE_TARNAME=\"newlib\" -DPACKAGE_VERSION=\"3.2.0\" -DPACKAGE_STRING=\"newlib\ 3.2.0\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -I. -I/home/sun/riscv-gnu-toolchain/riscv-newlib/newlib/libc/stdlib -fno-builtin -DHAVE_NANOSLEEP      -O2 -D_POSIX_MODE   -mcmodel=medlow -c -o lib_a-__adjust.o `test -f '__adjust.c' || echo '/home/sun/riscv-gnu-toolchain/riscv-newlib/newlib/libc/stdlib/'`__adjust.c
Assembler messages:
Fatal error: invalid -march= option: `rv64imafdv'
make[6]: *** [Makefile:632: lib_a-__adjust.o] Error 1
make[6]: Leaving directory '/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/libc/stdlib'
Making all in ctype
make[6]: Entering directory '/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/libc/ctype'
riscv64-unknown-elf-gcc -B/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/ -isystem /home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/targ-include -isystem /home/sun/riscv-gnu-toolchain/riscv-newlib/newlib/libc/include -B/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/libgloss/riscv64 -L/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/libgloss/libnosys -L/home/sun/riscv-gnu-toolchain/riscv-newlib/libgloss/riscv64    -DPACKAGE_NAME=\"newlib\" -DPACKAGE_TARNAME=\"newlib\" -DPACKAGE_VERSION=\"3.2.0\" -DPACKAGE_STRING=\"newlib\ 3.2.0\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -I. -I/home/sun/riscv-gnu-toolchain/riscv-newlib/newlib/libc/ctype -fno-builtin -DHAVE_NANOSLEEP      -O2 -D_POSIX_MODE   -mcmodel=medlow -c -o lib_a-ctype_.o `test -f 'ctype_.c' || echo '/home/sun/riscv-gnu-toolchain/riscv-newlib/newlib/libc/ctype/'`ctype_.c
Assembler messages:
Fatal error: invalid -march= option: `rv64imafdv'
make[6]: *** [Makefile:537: lib_a-ctype_.o] Error 1
make[6]: Leaving directory '/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/libc/ctype'
Making all in search
make[6]: Entering directory '/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/libc/search'
riscv64-unknown-elf-gcc -B/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/ -isystem /home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/targ-include -isystem /home/sun/riscv-gnu-toolchain/riscv-newlib/newlib/libc/include -B/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/libgloss/riscv64 -L/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/libgloss/libnosys -L/home/sun/riscv-gnu-toolchain/riscv-newlib/libgloss/riscv64    -DPACKAGE_NAME=\"newlib\" -DPACKAGE_TARNAME=\"newlib\" -DPACKAGE_VERSION=\"3.2.0\" -DPACKAGE_STRING=\"newlib\ 3.2.0\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -I. -I/home/sun/riscv-gnu-toolchain/riscv-newlib/newlib/libc/search -fno-builtin -DHAVE_NANOSLEEP      -O2 -D_POSIX_MODE   -mcmodel=medlow -c -o lib_a-bsearch.o `test -f 'bsearch.c' || echo '/home/sun/riscv-gnu-toolchain/riscv-newlib/newlib/libc/search/'`bsearch.c
Assembler messages:
Fatal error: invalid -march= option: `rv64imafdv'
make[6]: *** [Makefile:404: lib_a-bsearch.o] Error 1
make[6]: Leaving directory '/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/libc/search'
Making all in stdio
make[6]: Entering directory '/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/libc/stdio'
riscv64-unknown-elf-gcc -B/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/ -isystem /home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/targ-include -isystem /home/sun/riscv-gnu-toolchain/riscv-newlib/newlib/libc/include -B/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/libgloss/riscv64 -L/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/libgloss/libnosys -L/home/sun/riscv-gnu-toolchain/riscv-newlib/libgloss/riscv64    -DPACKAGE_NAME=\"newlib\" -DPACKAGE_TARNAME=\"newlib\" -DPACKAGE_VERSION=\"3.2.0\" -DPACKAGE_STRING=\"newlib\ 3.2.0\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -I. -I/home/sun/riscv-gnu-toolchain/riscv-newlib/newlib/libc/stdio -fno-builtin -DHAVE_NANOSLEEP      -O2 -D_POSIX_MODE   -mcmodel=medlow -c -o lib_a-fiprintf.o `test -f 'fiprintf.c' || echo '/home/sun/riscv-gnu-toolchain/riscv-newlib/newlib/libc/stdio/'`fiprintf.c
Assembler messages:
Fatal error: invalid -march= option: `rv64imafdv'
make[6]: *** [Makefile:820: lib_a-fiprintf.o] Error 1
make[6]: Leaving directory '/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/libc/stdio'
Making all in string
make[6]: Entering directory '/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/libc/string'
riscv64-unknown-elf-gcc -B/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/ -isystem /home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/targ-include -isystem /home/sun/riscv-gnu-toolchain/riscv-newlib/newlib/libc/include -B/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/libgloss/riscv64 -L/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/libgloss/libnosys -L/home/sun/riscv-gnu-toolchain/riscv-newlib/libgloss/riscv64    -DPACKAGE_NAME=\"newlib\" -DPACKAGE_TARNAME=\"newlib\" -DPACKAGE_VERSION=\"3.2.0\" -DPACKAGE_STRING=\"newlib\ 3.2.0\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -I. -I/home/sun/riscv-gnu-toolchain/riscv-newlib/newlib/libc/string -fno-builtin -DHAVE_NANOSLEEP      -O2 -D_POSIX_MODE   -mcmodel=medlow -c -o lib_a-bcopy.o `test -f 'bcopy.c' || echo '/home/sun/riscv-gnu-toolchain/riscv-newlib/newlib/libc/string/'`bcopy.c
Assembler messages:
Fatal error: invalid -march= option: `rv64imafdv'
make[6]: *** [Makefile:581: lib_a-bcopy.o] Error 1
make[6]: Leaving directory '/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/libc/string'
Making all in signal
make[6]: Entering directory '/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/libc/signal'
riscv64-unknown-elf-gcc -B/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/ -isystem /home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/targ-include -isystem /home/sun/riscv-gnu-toolchain/riscv-newlib/newlib/libc/include -B/home/sun/riscv-gnu- [... cut ...]
make[4]: Leaving directory '/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib'
make[3]: *** [Makefile:452: all] Error 2
make[3]: Leaving directory '/home/sun/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib'
make[2]: *** [Makefile:8492: all-target-newlib] Error 2
make[2]: Leaving directory '/home/sun/riscv-gnu-toolchain/build-newlib'
make[1]: *** [Makefile:879: all] Error 2
make[1]: Leaving directory '/home/sun/riscv-gnu-toolchain/build-newlib'
make: *** [Makefile:553: stamps/build-newlib] Error 2

jim-wilson, 2021-08-12:
Are you sure that you are on the rvv-intrinsic branch?  What does "git status" say?  Sometimes people switch branches and forget to update the submodules.

Do you have an existing toolchain in /home/sun/output/?  You can get odd build errors if you try to build a toolchain overtop another one that was configured differently.  This is mentioned in the Troubleshooting section of the README.md file.  I suggest using a different prefix for every compiler build to avoid this problem.

vigoss30611, 2021-08-12:
> Are you sure that you are on the rvv-intrinsic branch? What does "git status" say? Sometimes people switch branches and forget to update the submodules.
>
> Do you have an existing toolchain in [...]

I am sure on the rvv-intrinsic branch.
Is my command right? ./configure --prefix=/home/sun/output/ --with-arch=rv64imafdv --with-abi=lp64 
Because when I change --with-arch=rv64imafdv to --with-arch=rv64imafdc, it can run right,

jim-wilson, 2021-08-12:
Yes, your configure options are OK.  The most likely causes of trouble are the 2 issues I already mentioned.

Another possible problem is that you might accidentally be running the native x86_64 assembler instead of the riscv64-elf assembler.  The assembler -march option looks like it might be from the native x86_64 assembler.  To check this, try compiling a hello world program with your riscv64-unknown-elf-gcc.  If you get a link error, then it is OK, as you haven't built libc yet.  If you get an assembler error, then there is something wrong, and it might be an environment variable like PATH, or maybe something more obscure like COMPILER_PATH.  Or maybe corruption of your install tree.  Try adding a -v option to the hello world program compile.  That will show you what the driver is doing, and will show exactly which assembler it is trying to use.  It should be using a riscv assembler from your toolchain install tree.  If it is using /usr/bin/as then something is clearly wrong.  You should also try looking at your install tree to make sure it looks OK.  There should be a iscv64-unknown-elf/bin/as binary there for the compiler to use.

TommyMurphyTM1234, 2022-09-29:
No update/activity in over a year and, these days, the branch on which V extension support is being implemented/enhanced is `rvv-next` (https://github.com/riscv-collab/riscv-gnu-toolchain/tree/rvv-next) and `rvv-intrinsic` no longer exists.

closed 2022-09-29 by TommyMurphyTM1234