riscv-collab/riscv-gnu-toolchain #943: Cannot find -lz while cross-compiling risc-v gnu toolchain

Sairamtcs, 2021-08-12:
Hi,
I'm cross-compiling risc-v gnu toolchain for RISCV linux target, during the process it is cross compiled the source files but while linking the linker is unable to locate zlib library, Please look the following for reference.

riscv64-unknown-linux-gnu-g++ -no-pie   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc  gcov.o json.o \
	hash-table.o ggc-none.o libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a  -lz -o gcov
/opt/riscv64/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find -lz
collect2: error: ld returned 1 exit status
Makefile:2932: recipe for target 'gcov' failed
make[2]: *** [gcov] Error 1
make[2]: *** Waiting for unfinished jobs....
rm fsf-funding.pod gcov.pod gpl.pod cpp.pod gfdl.pod gcc.pod gcov-dump.pod gcov-tool.pod
make[2]: Leaving directory '/home/tcs/riscv-gnu-toolchain/build-gcc-newlib-stage1/gcc'
Makefile:4324: recipe for target 'all-gcc' failed
make[1]: *** [all-gcc] Error 2
make[1]: Leaving directory '/home/tcs/riscv-gnu-toolchain/build-gcc-newlib-stage1'
Makefile:428: recipe for target 'stamps/build-gcc-newlib-stage1' failed
make: *** [stamps/build-gcc-newlib-stage1] Error 2

jim-wilson, 2021-08-12:
So you are trying to use riscv-gnu-toolchain to build a riscv64 native linux compiler.  Don't.  It isn't designed to do that.  You should instead do a sysroot style build.  Create a sysroot from your target system.  A sysroot contains everything that the compiler needs, which is generally /usr/include, /lib, /lib* if they exist, /usr/lib, and /usr/lib* if they exist.  Since the sysroot includes a glibc build, you only need binutils and gcc builds.  First do a --build=x86_64-linux --host=x86_64-linux --target=riscv64-linux build adding a --sysroot= option pointing at your sysroot.  Then do a --build=x86_64-linux --host=riscv64-linux --target=riscv64-linux cross native build using the cross build you just built again adding a sysroot= option pointing at the sysroot.  You can then copy this to the target and it will mostly work except for some path confusion, so you may need to create the same install tree on the target as on the build machine.  Then do a native build on the target to get the paths right.

I used riscv64-linux here as a shorthand, but the target value you use here should match the target system or you may run into problems, and different linux distros use different targets here, so check the gcc.  If the target is debian/ubuntu they you need --enable-multiarch, and probably non-upstream patches to make that work.  OpenEmbedded and Fedora are easier targets.

The reason why your build failed is because riscv-gnu-toolchain only builds gcc libraries and glibc.  zlib is a separate package that isn't built by riscv-gnu-toolchain.  If you do build zlib, you will find that other packages are missing too.  If you want a cross native toolchain build this way, you essentially need to build an entire linux distro.  And if you are doing that, then it is much easier to use one that already exists like OpenEmbedded or gentoo than to try to create your own.

TommyMurphyTM1234, 2022-09-27:
Incorrect approach to building toolchain. Not an issue with the RISC-V GCC toolchain.

closed 2022-09-27 by TommyMurphyTM1234