riscv-collab/riscv-gnu-toolchain #945: How to build toolchain for V, B, and ZFH extensions

jrahmeh, 2021-08-16:
Is there a way to build the toolchain with support for the V, B, and ZFH extensions?

jim-wilson, 2021-08-16:
There is no easy way to do this.  The zba/zbb/zbc binutils support is upstream on the master branch.  The rest of the B support is on a branch in riscv-binutils-gdb.  The vector and zfh binutils support is upstream on the integration branch which rebases from master.  So you can get all 3 if you only need zba/zbb/zbc

But the GCC support is not upstream, and B and V are on different branches, with zfh on the V branch.  So you have to merge patches together which is tricky.

You might be able to get this by downloading a toolchain binary release from the SiFive web site, but I'm not sure what exactly is in our last public release, and even if it has the extensions you want, they may not be the right versions of those extensions.

Unfortunately, things are in an unmanagable state at the moment.  Too many draft extensions, changing too often, with patches coming from too many different organizations, and too many restrictions on what we can put upstream.  Maybe things will settle down next year after a bunch of extensions get ratified.

jrahmeh, 2021-08-18:
Hi Jim, thanks for the explanation.  Where can I find the binutils integration branch?

jim-wilson, 2021-08-23:
The upstream binutils branch is users/riscv/binutils-integration-branch.

jrahmeh, 2021-08-30:
Hi Jim, thank you very much for your suggestions.  They worked.  We are now able to compile with V, B, and ZFH.

closed 2021-08-30 by jrahmeh