riscv-collab/riscv-gnu-toolchain #946: riscv-glibc can't clone

amithmath, 2021-08-23:
I can't clone riscv-glibc and riscv-newlib. Please suggest me solutions to fix....

jim-wilson, 2021-08-23:
What exact command did you type and what exact error did you get?  Are you trying to clone them directly?  Are you using git submodule update --init --recursive?  Or are you using make and letting it do the submodule updates?  Or doing something else?

Some people have reported that using make to do the submodule update doesn't work for them, but running git submodule update --init --recursive should always work (unless there are network problems).

Note that glibc is upstream, and a direct clone of riscv-glibc won't give you anything useful.  You need to clone riscv-gnu-toolchain and then update submodules.  Or clone FSF glibc if you want to clone glibc directly.

Both of the glibc and newlib submodules are coming from sourceware.org.  A failure to access them could be a temporary problem with the sourceware.org site.  Ot there could be a firewall blocking access to sourceware.org.  Try visiting the web site to see if you can reach it.  If you can't reach it, try waiting a few hours and trying again.

amithmath, 2021-08-23:
I did clone riscv-gnu-toolchain and  git submodule update --init --recursive, then also glibc is not cloning.

jim-wilson, 2021-08-24:
That just worked fine for me.  Can you reach the sourceware.org web site?  The glibc sources are at https://sourceware.org/git/?p=glibc.git;a=summary you can try pulling that with the git commands listed near the top, maybe a firewall is blocking some access methods.  If one works but the others don't, then you can use a url ... insteadof stanza in your ~/.gitconfig file to force use of the protocol that works.

amithmath, 2021-08-24:
Yes, if I individually clone from sourceware.org it works...

closed 2021-08-24 by amithmath

sahooora, 2021-11-03:
Hi,

I've faced the same problem.

My machine is Linux Mint 19.

After `make linux` I've got this error:
`cd /home/sara/riscv/riscv-gnu-toolchain && \
flock `git rev-parse --git-dir`/config git submodule init /home/sara/riscv/riscv-gnu-toolchain/riscv-glibc/ && \
flock `git rev-parse --git-dir`/config git submodule update /home/sara/riscv/riscv-gnu-toolchain/riscv-glibc/
Cloning into '/home/sara/riscv/riscv-gnu-toolchain/riscv-glibc'...
fatal: unable to connect to sourceware.org:
sourceware.org[0: 8.43.85.97]: errno=Connection refused
sourceware.org[1: 2620:52:3:1:0:246e:9693:128c]: errno=Network is unreachable

fatal: clone of 'git://sourceware.org/git/glibc.git' into submodule path '/home/sara/riscv/riscv-gnu-toolchain/riscv-glibc' failed
Failed to clone 'riscv-glibc'. Retry scheduled
Cloning into '/home/sara/riscv/riscv-gnu-toolchain/riscv-glibc'...
fatal: unable to connect to sourceware.org:
sourceware.org[0: 8.43.85.97]: errno=Connection refused
sourceware.org[1: 2620:52:3:1:0:246e:9693:128c]: errno=Network is unreachable

fatal: clone of 'git://sourceware.org/git/glibc.git' into submodule path '/home/sara/riscv/riscv-gnu-toolchain/riscv-glibc' failed
Failed to clone 'riscv-glibc' a second time, aborting
Makefile:247: recipe for target '/home/sara/riscv/riscv-gnu-toolchain/riscv-glibc/.git' failed
make: *** [/home/sara/riscv/riscv-gnu-toolchain/riscv-glibc/.git] Error 1
` 

I've also tried `git submodule update --init --recursive` but the same error was occurred.

Thanks

f-riente, 2022-03-09:
Hi,

I'm getting a similar error again when running:

`make linux`

On Ubuntu 20. I have checked and I can reach the sourceware website from the same machine.
I have also tried cloning directly riscv-glibc in another folder using the command at the top of https://sourceware.org/git/?p=glibc.git;a=summary

The error I get in both cases is:
```
flock `git rev-parse --git-dir`/config git submodule init /home/riente/Documents/riscv-gnu-toolchain/riscv-glibc/ && \
flock `git rev-parse --git-dir`/config git submodule update /home/riente/Documents/riscv-gnu-toolchain/riscv-glibc/
Cloning into '/home/user/Documents/riscv-gnu-toolchain/riscv-glibc'...
fatal: unable to connect to sourceware.org:
sourceware.org[0: 8.43.85.97]: errno=Connection refused
sourceware.org[1: 2620:52:3:1:0:246e:9693:128c]: errno=Network is unreachable

fatal: clone of 'git://sourceware.org/git/glibc.git' into submodule path '/home/riente/Documents/riscv-gnu-toolchain/riscv-glibc' failed
Failed to clone 'riscv-glibc'. Retry scheduled
Cloning into '/home/user/Documents/riscv-gnu-toolchain/riscv-glibc'...
fatal: unable to connect to sourceware.org:
sourceware.org[0: 8.43.85.97]: errno=Connection refused
sourceware.org[1: 2620:52:3:1:0:246e:9693:128c]: errno=Network is unreachable

fatal: clone of 'git://sourceware.org/git/glibc.git' into submodule path '/home/user/Documents/riscv-gnu-toolchain/riscv-glibc' failed
Failed to clone 'riscv-glibc' a second time, aborting
```
Any suggestions?

Thanks

jim-wilson, 2022-03-10:
Check connectivity to sourceware.org.  Can you view it in a browser? 
[https://sourceware.org/](https://sourceware.org/)
Does a direct git command work?  Does it matter if you use git or https?
```
    git clone git://sourceware.org/git/glibc.git
    git clone https://sourceware.org/git/glibc.git
```

If you are behind a corporate or national firewall that allows some connections but not others, then you can use a git feature to change the connection method.  For instance, in my ~/.gitconfig file I have
```
[url "ssh://git@github.com"]
	insteadOf = https://github.com
	insteadOf = http://github.com
	insteadOf = git://github.com
```
This ensures that I always use ssh when connecting to github.com.  In your case, you need to find the connection method that works (https?) and then force use of that instead of the other possibilities for the sourceware.org site.  So try something like this
```
[url "https://git@sourceware.org"]
        insteadOf = ssh://sourceware.org
        insteadOf = http://sourceware.org
        insteadOf = git://sourceware.org
```

jim-wilson, 2022-03-10:
I forgot to drop the git@, that should be just
```
[url "https://sourceware.org"]
```

f-riente, 2022-03-10:
Yes, I was able to visit the page from the browser.

I don't really know what was the problem, but today, from the same computer, same network the recursive clone of the `riscv-glibc` modules was working.

f-riente, 2022-03-11:
Dear @jim-wilson ,

thanks for your answers. It worked out by replacing git:// with https://
The error appears only with `git://sourceware.org/git/glibc.git`