riscv-collab/riscv-gnu-toolchain #947: What is the meaning of this opcode?

nikita890-art, 2021-08-24:
mod     rd rs1 rs2 31..25=1  14..12=0 6..2=0x1A 1..0=3

I am trying to understand what exactly this sentence means.
I found this line here,
https://nitish2112.github.io/post/adding-instruction-riscv/

what is the purpose of this line? 
What is "14..12=0" , "6..2=0x1A" and "1..0=3" ?

jim-wilson, 2021-08-24:
This looks like something taken from riscv/riscv-opcodes.  The answer will be found there.  But it looks like they are specifying the arguments, and the values to put in every opcode field that isn't an arg field.  E.g. bits 31 to 25 of the opcode are set to 00001.

TommyMurphyTM1234, 2022-09-27:
No update/activity for over a year and it seems to be a simple misinterpretation or confusion over meaning of the linked document which is not part of this repo. Not a RISC-V GCC toolchain issue.

closed 2022-09-27 by TommyMurphyTM1234