riscv-collab/riscv-gnu-toolchain #949: LD does an incorrect relaxation?

HAKarlsson, 2021-08-25:
I am writing a trap handler and detected that the following assembly snippet
```
      1:auipc   t0,%pcrel_hi(trap_table)
        slli    t1,a0,LOG_REGBYTES
        add     t1,t0,t1
        ld      t0,%pcrel_lo(1b)(t1)
        jalr    ra,t0
```
compiles to (objdump of the object file)
```
  ba:	00000297          	auipc	t0,0x0
  be:	00351313          	slli	t1,a0,0x3
  c2:	9316                	add	t1,t1,t0
  c4:	00033283          	ld	t0,0(t1)
  c8:	000280e7          	jalr	t0 # ba <.L1^B1>
```
and after linkage with `ld` 
```
    800000e8:	00351313          	slli	t1,a0,0x3
    800000ec:	9316                	add	t1,t1,t0
    800000ee:	c001b283          	ld	t0,-1024(gp) # 80000190 <trap_table>
    800000f2:	000280e7          	jalr	t0
```
so the load ignores the offset `t1` to my trap table. So basically, the result is `trap_table[0](...)` instead of the expected `trap_table[a0](...)`.

Here are the compiler and ld versions.
```
riscv64-unknown-elf-gcc (GCC) 11.1.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

GNU ld (GNU Binutils) 2.37
Copyright (C) 2021 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
```

aswaterman, 2021-08-25:
It is defined that auipc/load pairs can be relaxed into gp-relative load pairs, so this is expected behavior even though it is undesirable in this case. In effect, there is an ABI constraint that prohibits your otherwise-sensible code sequence.

You can try wrapping this sequence in .option norelax (see https://github.com/bminor/glibc/blob/master/sysdeps/riscv/start.S#L72) or alternatively rewrite as lla + add + ld, which will relax correctly into addi + add + ld. Both will end up performing about the same.

jim-wilson, 2021-08-25:
On Wed, Aug 25, 2021 at 9:06 AM Henrik 昭 Karlsson ***@***.***>
wrote:

> I am writing a trap handler and detected that the following assembly
> snippet
>
>       1:auipc   t0,%pcrel_hi(trap_table)
>         slli    t1,a0,LOG_REGBYTES
>         add     t1,t0,t1
>         ld      t0,%pcrel_lo(1b)(t1)
>         jalr    ra,t0
>
>
We need a new relocation and a new linker relaxation to make this work.  I
have a comment about this at
https://github.com/riscv/riscv-code-speed-optimization/blob/main/projects/gcc-optimizations.adoc#eliminate-extra-array-adds

Jim