riscv-collab/riscv-gnu-toolchain #951: How to build lli (LLVM JIT) for RISCV?

Joejiong, 2021-08-28:
Hi everyone,

I am trying to build a JIT for the RISC-V platform, but I could not figure
out how. Could you provide some hints? Great thanks!

Initially, I found that LLVM has RISCV backend, and I compiled it following
the instructions here <https://github.com/lowRISC/riscv-llvm>. I got a
cross-compiler version of clang (running on x86 and the generated binary
runs on RISC-V), but the lli is still a X86 tool.

Then, I tried to build LLVM with riscv-gnu-toolchain
<https://github.com/riscv/riscv-gnu-toolchain>, and I got many tools such
as llc working on RISC-V properly. However, the lli still could not work
properly. When I tried to process the LLVM bytecode file (.ll file) on
RISC-V using lli, I got the error:
*         error creating EE: No available targets are compatible with this
triple.*
Furthermore, when I typed: ./lli --version, I got:

Default target:
x86_64-unknown-linux-gnu  
Host CPU: (unknown)*

But when I typed: llc -version (llc works fine), I got:
  DEBUG build with assertions.
  Default target: x86_64-unknown-linux-gnu
  Host CPU: (unknown)
  Registered Targets:
    riscv32 - 32-bit RISC-V
    riscv64 - 64-bit RISC-V
    x86     - 32-bit X86: Pentium-Pro and above
    x86-64  - 64-bit X86: EM64T and AMD64
It seems that the lli tool does not support RISC-V target.

Could anyone give me some hints? Thank you so much!!

Best,

jim-wilson, 2021-08-28:
> Initially, I found that LLVM has RISCV backend, and I compiled it following
> the instructions here https://github.com/lowRISC/riscv-llvm. I got a

This is obsolete.  Note that the last commit was 2018.  Use llvm.org instead.

This isn't the best place to ask llvm questions.  Try an llvm or RISC-V International mailing list.

TommyMurphyTM1234, 2022-09-27:
LLVM questions - not an issue with the RISC-V GCC toolchain.

closed 2022-09-27 by TommyMurphyTM1234