riscv-collab/riscv-gnu-toolchain #952: Assembler Options

muhammadaamir813, 2021-09-01:
After executing following command : riscv64-unknown-elf-gcc -O2 -o Hello_World Hello_World.c
I got the result
Assembler messages:
Fatal error: invalid -march= option: `rv64imafdc'

jim-wilson, 2021-09-01:
You might accidentally be running the native (x86?) assembler.  Try adding a -v option and look to see which assembler the gcc driver is using.  The assembler it should be using is $prefix/riscv64-unknown-elf/bin/as.   Check to make sure that this directory exists and that there is an assembler binary inside there.

riscv64-unknown-elf-gcc --print-search-dirs will show you the path that the compiler uses to find its subprograms, including the assembler.  The directory that holds the assembler should be on this list of dirs.

TommyMurphyTM1234, 2022-09-27:
No update/activity in over a year and nothing that is obviously an issue with the RISC-V GCC toolchain itself.

closed 2022-09-27 by TommyMurphyTM1234