riscv-collab/riscv-gnu-toolchain #953: Issue building anything since binutils 2.37 bump

fwsGonzo, 2021-09-03:
Latest working commit: c98135b7edfa5356f92c6146c1622f683e90d177 (2 commits back)

I cannot build anything since the bump. I don't know what's wrong but it wonder if it could just be a missing config variable.
I only tested newlibs and Linux. They each don't build for RV32G and RV64G.

Ubuntu 20.04.3 LTS
```
$ uname -a
Linux 5.4.0-80-generic #90-Ubuntu SMP Fri Jul 9 22:49:44 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

jim-wilson, 2021-09-03:
I can only help if I can reproduce what you are going.  But it isn't clear exactly what you are doing, or exactly what error you are getting.  This report is confusing.

I see a use of -print-multi-os-directory in the newlib libtool.m4 shell script, but it is protected by
```
if test "$GCC" = yes; then
```
so it is not clear how using clang could generate that error.  And not clear if you are getting this error from the newlib build.  it is possible that this option could be used in other places.

For the glibc error, does the directory for the flock exist?  If it doesn't, why not?  Did you configure this dir as --prefix?  Why are you using install_root=?  We don't support that or use that in riscv-gnu-toolchain.

Just guessing, I would ask if you are trying to build using a --prefix that already contains a compiler, which is a different version, or different configuration.  This is known to cause problems, and is mentioned in the troubleshooting section.  You should always use a different and unique --prefix for every toolchain build to avoid this problem.

I don't know if riscv-gnu-toolchain can be built with clang.  I don't use clang.

fwsGonzo, 2021-09-03:
I use bog-standard configure with --pregix=$HOME/riscv - I've had a quadruple of newlib 32g, newlib64, linux 32g, linux 64g for a while.

So, before I waste any more of your time I want to try building with GCC. And as you said, maybe a unique prefix too. Thanks for that tip.

fwsGonzo, 2021-09-05:
Hmm, something has changed and now it doesn't accept RV32G anymore:
```
$ ./configure --prefix=$HOME/riscv --with-arch=rv32g --with-abi=ilp32d
```
Results in building for 10 minutes and then a bunch of errors about invalid -march=rv32g for binutils.

At this point I'm just going one by one commit back until it works again. What happened?

fwsGonzo, 2021-09-05:
Alright, so this commit works: 9f4504dd25587f4a7e532e5e77ec8195118c5b98
Not very far back. I was able to build rv32g newlib with it.

jim-wilson, 2021-09-05:
binutils 2.37 is working for everyone else.  It may be a problem with your environment.  How exactly does the build fail?  You didn't show the error message or a build log or anything I can use to determine what your problem is.

The invalid -march error suggests that you are accidentally using the native x86 assembler during the build instead of the cross riscv assembler.

fwsGonzo, 2021-09-11:
```
$ git reset --hard origin/master
HEAD is now at 1a36b5d Merge pull request #959 from riscv-software-src/fix-repo-url
$ git submodule update --init --depth 1
remote: Total 0 (delta 0), reused 0 (delta 0)
error: Server does not allow request for unadvertised object 553032db17440f8de011390e5a1cfddd13751b0b
Fetched in submodule path 'qemu', but it did not contain 553032db17440f8de011390e5a1cfddd13751b0b. Direct fetching of that commit failed.
```

jim-wilson, 2021-09-11:
This is a problem on the qemu side.  Looks like the qemu-5.2 branch and tag was accidentally deleted and then later restored.  I was able to reproduce this problem when I looked at it, but I had to do some other chores first, and by the time I got back to this the qemu-5.2 branch and tag had reappeared again.  Shrug.

fwsGonzo, 2021-09-11:
That's timing for you. Sorry to bother with these small problems. I am in the process of trying to build RV128I support, and I find that there isn't any toolchain support right. And no ELFCLASS for 128-bit anyway. So I am just building 64-bit binaries manually and then hand-editing a few things. :man_shrugging: 

I'll close this now. Hopefully no more building issues.

closed 2021-09-11 by fwsGonzo