riscv-collab/riscv-gnu-toolchain #954: Difference between native GCC and RISC-V GCC

fwsGonzo, 2021-09-04:
Hey, there's a difference that I have trouble explaining. I have an idea that it's related to how RISC-V is sign-extending differently than amd64. Still, have a look at this:

```
#include <stdio.h>

int main() {
	volatile char a = 0xff;
	volatile char b = 1;
	printf("%d ", (int)(a + b));
	a = -1;
	printf("%d\n", (int)(a + b));
}
```

```
$ ./emulator fail
256 256
```
An emulator I wrote prints 256 for each, as if adding 2 integers. I always think the fault is mine.

```
$ gcc -static fail.c -o native_fail
 ./native_fail 
0 0
```
As I said above, it prints 0 here.

Is this correct behavior?

jim-wilson, 2021-09-05:
The ISO C standard says "The implementation shall define char to have the same range, representation, and behavior as either signed char or unsigned char."  So anything using "char" is ambiguous, as it may be either a signed or unsigned type.  On x86, char is signed by default.  On RISC-V, char is unsigned by default.  ARM also has char unsigned by default.

fwsGonzo, 2021-09-05:
Thanks for the explanation. I did not expect that!

closed 2021-09-05 by fwsGonzo