riscv-collab/riscv-gnu-toolchain #955: while(1) loop with malloc and free for bare-metal systems

Kian75, 2021-09-05:
Hi I am recently working on RiscV and I have written a C program for Ibex core which should happen inside a while(1) loop.. In this while(1) loop I have used malloc  and free functions to manage memory... and I used _sbrk for bare-metal systems... but my problem is that first iteration is ok and code reaches to the end and I see the LED lights up but in second iteration of while(1) it stuck right before first malloc in my code (meaning it doesn't reach to the end of the code and LED doesn't light up) I freed all of the mallocs so I don't think there is any leak ... what can be the problem? I am really confused...
This is the _sbrk I use:

`void*
_sbrk(incr)
int incr;
{
    extern char   end; /* Set by linker.  */
    static char* heap_end;
    char* prev_heap_end;

    if (heap_end == 0)
        heap_end = &end;

    prev_heap_end = heap_end;
    heap_end += incr;

    return (void*)prev_heap_end;
}`

and my linker file:

`OUTPUT_ARCH(riscv)

/* required to correctly link newlib */
GROUP( -lc -lgcc -lsupc++ )

SEARCH_DIR(.)
__DYNAMIC  =  0;

MEMORY
{
    rom         : ORIGIN = 0x00000000, LENGTH = 0x36800 /* 218 kB */
    stack       : ORIGIN = 0x00036800, LENGTH = 0x4F000 /* 320 kB */
}

/* Stack information variables */
_min_stack      = 0x2000;   /* 8K - minimum stack space to reserve */
_stack_len     = LENGTH(stack);
_stack_start   = ORIGIN(stack) + LENGTH(stack);

/* We have to align each sector to word boundaries as our current s19->slm
 * conversion scripts are not able to handle non-word aligned sections. */

SECTIONS
{
    .vectors :
    {
        . = ALIGN(4);
        KEEP(*(.vectors))
    } > rom

    .text : {
        . = ALIGN(4);
        _stext = .;
        *(.text)
        *(.text.*)
        _etext  =  .;
        __CTOR_LIST__ = .;
        LONG((__CTOR_END__ - __CTOR_LIST__) / 4 - 2)
        *(.ctors)
        LONG(0)
        __CTOR_END__ = .;
        __DTOR_LIST__ = .;
        LONG((__DTOR_END__ - __DTOR_LIST__) / 4 - 2)
        *(.dtors)
        LONG(0)
        __DTOR_END__ = .;
        *(.lit)
        *(.shdata)
        . = ALIGN(4);
        _endtext = .;
    }  > rom

    .rodata : {
        . = ALIGN(4);
        *(.rodata);
        *(.rodata.*)
    } > rom

    .shbss :
    {
        . = ALIGN(4);
        *(.shbss)
    } > rom

    .data : {
        . = ALIGN(4);
        sdata  =  .;
        _sdata  =  .;
        *(.data);
        *(.data.*)
        edata  =  .;
        _edata  =  .;
    } > rom

    .bss :
    {
        . = ALIGN(4);
        _bss_start = .;
        *(.bss)
        *(.bss.*)
        *(.sbss)
        *(.sbss.*)
        *(COMMON)
        _bss_end = .;
    } > rom
    end = .;

    /* ensure there is enough room for stack */
    .stack (NOLOAD): {
        . = ALIGN(4);
        . = . + _min_stack ;
        . = ALIGN(4);
        stack = . ;
        _stack = . ;
    } > stack

    .stab  0 (NOLOAD) :
    {
        [ .stab ]
    }

    .stabstr  0 (NOLOAD) :
    {
        [ .stabstr ]
    }
}
`

jim-wilson, 2021-09-05:
I don't think that there is much we can do to solve this.  We don't have your code or your target.

I would suggest that you check the sp and heap_end values.  Maybe they are colliding.  Low end embedded targets usually don't have much memory, and malloc is pretty heavy weight, it might be using more memory than you realize.

TommyMurphyTM1234, 2021-09-23:
What if you use newlib's own/default implementation of `_sbrk()` rather than providing your own?
Does that work?
If it does then you can then compare it with your implementation.
Of course the issue could be something completely unrelated to `_sbrk()`.
As Jim says it's difficult to comment authoritatively in the absence of the relevant target and complete source code.
(And I agree with the implication of Jim's comment - that heap based memory [de]allocation in a memory contstrained bare metal embedded environment really needs strong justification and maybe a very robust custom allocator/deallocator implementation).

closed 2022-09-27 by TommyMurphyTM1234