riscv-collab/riscv-gnu-toolchain #956: march rv32imafc does not use RV32F instructions!!!

apivovarov, 2021-09-05:
I noticed that gcc does NOT generate assembly code with `RV32F` instructions for `rv32imafc` march.

My test program - `test.c`
```
float test_func(float c) {
    float k = c + 3.1456;
    float n = k * 0.55;
    float z = n / 12.55;
    return z;
}
```
Compile it with `march` and `mabi` flags for F extension.
```
riscv64-unknown-linux-gnu-gcc \
-march=rv32imafc \
-mabi=ilp32f \
-S test.c -o test.s
```
Resulting `test.s` uses soft-float Instructions.
```
        .globl  __adddf3
        .globl  __truncdfsf2
        .globl  __muldf3
        .globl  __divdf3
```
But I expected it to use RV32F instructions instead, such as `FADD.S`, `FMUL.S`, `FDIV.S`

kito-cheng, 2021-09-05:
Because floating point literal is default to double precision in C, that result every operation in `test_func` is double precision operation...

So here is two solutions for your case:
1. Add `f` suffix to floating point literal, that can let those floating point literal become single precision .
2. Use `-fsingle-precision-constant` to force every floating point literal become single precision, but this is NOT recommend way, since it's result different behavior with the standard C/C++.

apivovarov, 2021-09-05:
Thank you, Kito!

closed 2021-09-05 by apivovarov