riscv-collab/riscv-gnu-toolchain #957: cannot find vlseg3e8_v_u8m1() and vsseg3e8_v_u8m1()

figozhang, 2021-09-07:
i wrote some test code for RVV intrinsic, but it looks like some functions cannot find in GCC.
i am using riscv-gnu-toolchain repo and "rvv-intrinsic" branch to build the GCC.

```
riscv64-unknown-elf-gcc -o test  rgb24_bgr24_test_c.o 
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: rgb24_bgr24_test_c.o: in function `rgb24_bgr24_neon_intr':
/home/rlk/rlk/riscv_trainning/vectors/example_rgb24_bgr24_intrinsic/rgb24_bgr24_test.c:30: undefined reference to `vlseg3e8_v_u8m1'
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: /home/rlk/rlk/riscv_trainning/vectors/example_rgb24_bgr24_intrinsic/rgb24_bgr24_test.c:36: undefined reference to `vsseg3e8_v_u8m1'
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: rgb24_bgr24_test_c.o: in function `main':
/home/rlk/rlk/riscv_trainning/vectors/example_rgb24_bgr24_intrinsic/rgb24_bgr24_test.c:70: undefined reference to `rgb24_bgr24_intr'
collect2: error: ld returned 1 exit status
make: *** [Makefile:19: test] Error 1
```

here is my test code:
```
static void rgb24_bgr24_neon_intr(unsigned char *src, unsigned char *dst, unsigned long count)
{
	unsigned long i;
	size_t vl;
	vuint8m1_t v0;
	vuint8m1_t v1;
	vuint8m1_t v2;
	vuint8m1_t v4;

	while(count) {
		vl = vsetvl_e8m1(count);
		vlseg3e8_v_u8m1 (&v0, &v1, &v2, src, vl);
		
		v4 = vmv_v_v_u8m1(v0, vl);
		v0 = vmv_v_v_u8m1(v2, vl);
		v2 = vmv_v_v_u8m1(v4, vl);
		
		vsseg3e8_v_u8m1 (dst, &v0, &v1, &v2, vl);

		count -= 48;
		src += 48;
		dst += 48;
	}
}

````

anyone can tell me which branch can support those two intrinsic funtions?

rjiejie, 2021-09-07:
you should compile with vector extension in command line option '-march' like
-march=rv64gcv

figozhang, 2021-09-07:
i have using the "-march=rv64gcv"

rjiejie, 2021-09-07:
Got it,
I think you need to patch my PR now :)
https://github.com/riscv/riscv-gcc/pull/291

it's under review :)

figozhang, 2021-09-07:
cool, i can try it.

but, i change my code, it can compile now, but i disassemble，i found the assemble code is not correct.

```
static void rgb24_bgr24_intr(unsigned char *src, unsigned char *dst, unsigned long count)
{
	size_t vl;
	vuint8m1x3_t rgb;
	vuint8m1x3_t bgr;

	while(count) {
		vl = vsetvl_e8m1(count);
		rgb = vlseg3e8_v_u8m1x3(src, vl);

		bgr.__val[2] = vmv_v_v_u8m1(rgb.__val[0], vl);
		bgr.__val[0] = vmv_v_v_u8m1(rgb.__val[2], vl);
		bgr.__val[1] = vmv_v_v_u8m1(rgb.__val[1], vl);
		
		vsseg3e8_v_u8m1x3(dst, bgr, vl);

		count -= 3 * vl;
		src += 3 * vl;
		dst += 3 * vl;
	}
}

```

figozhang, 2021-09-07:
i copy your riscv_vector.h file from your PR, but compile my code fail

```
rlk@master:example_rgb24_bgr24_intrinsic$ make
riscv64-unknown-elf-gcc -g -march=rv64gcv --static -c rgb24_bgr24_test.c -o rgb24_bgr24_test_c.o
In file included from rgb24_bgr24_test.c:6:
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/include/riscv_vector.h: In function '_RVV_INT_WLMUL_ITERATOR':
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/include/riscv_vector.h:246:1: error: expected declaration specifiers before '_RVV_FLOAT_WLMUL_ITERATOR'
  246 | _RVV_FLOAT_WLMUL_ITERATOR (_RVVFLOAT_VECTOR_INSERT)
      | ^~~~~~~~~~~~~~~~~~~~~~~~~
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/include/riscv_vector.h:364:1: error: expected declaration specifiers before '__extension__'
  364 | _RVV_INT_ITERATOR (_RVVINTLD_FF)
      | ^~~~~~~~~~~~~~~~~
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/include/riscv_vector.h:364:1: error: expected declaration specifiers before '__extension__'
  364 | _RVV_INT_ITERATOR (_RVVINTLD_FF)
      | ^~~~~~~~~~~~~~~~~
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/include/riscv_vector.h:364:1: error: expected declaration specifiers before '__extension__'
  364 | _RVV_INT_ITERATOR (_RVVINTLD_FF)
      | ^~~~~~~~~~~~~~~~~
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/include/riscv_vector.h:364:1: error: expected declaration specifiers before '__extension__'
  364 | _RVV_INT_ITERATOR (_RVVINTLD_FF)
      | ^~~~~~~~~~~~~~~~~
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/include/riscv_vector.h:364:1: error: expected declaration specifiers before '__extension__'
  364 | _RVV_INT_ITERATOR (_RVVINTLD_FF)
      | ^~~~~~~~~~~~~~~~~
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/include/riscv_vector.h:364:1: error: expected declaration specifiers before '__extension__'
  364 | _RVV_INT_ITERATOR (_RVVINTLD_FF)
      | ^~~~~~~~~~~~~~~~~

```

rjiejie, 2021-09-07:
you should use manipulation interface for tuple 

> cool, i can try it.
>
> but, i change my code, it can compile now, but i disassemble，i found the assemble code is not correct.
>
> ```
> static void rgb24_bgr24_intr(unsigned char *src, unsigned char *dst, [...]

you should use manipulation interface for tuple as:

vget_u8m1x3_u8m1 (rgb, 0),

if you want to pack them to tuple type, you should use create interface as:

bgr = vcreate_u8m1x3 (vget_u8m1x3_u8m1 (rgb, 2), vget_u8m1x3_u8m1 (rgb, 1), vget_u8m1x3_u8m1 (rgb, 0),);

TommyMurphyTM1234, 2022-09-27:
No update/activity for over a year and seems to be confusion/questions about how to properly build the toolchain with vector support as opposed to being issues with the RISC-V GCC toolchain or build process itself.

closed 2022-09-27 by TommyMurphyTM1234