riscv-collab/riscv-gnu-toolchain #958: riscv-gcc repository is unavailable

IdanHo, 2021-09-09:
It looks like the riscv-gcc submodule in this repository was deleted/is no longer public for some reason

lazyparser, 2021-09-09:
> It looks like the riscv-gcc submodule in this repository was deleted/is no longer public for some reason

the repos under github/riscv had been separated into several different groups, just like the repo itself.

Seems like the submodules had been (wrongly?) move to different groups.

e.g. for gcc, try:
https://github.com/riscv-collab/riscv-gcc

jrpetrus, 2021-09-09:
I have this problem, too!

jim-wilson, 2021-09-09:
RISC-V International is reorganizing their github repo, and this accidentally broke riscv-gnu-toolchain.  riscv-gnu-toolchain is in riscv-software-src and riscv-gcc is in riscv-collab which breaks the ../ in the gitmodules file.  I sent email to them yesterday to discuss solutions.  The tentative plan is to move riscv-gnu-toolchain again to riscv-collab which will allow the ../ to work again.

Alternative solution is to change the .gitmodules file to use URLs instead of relative paths, but I can't go back in time and fix old versions, so I would prefer moving the repo.  Fixing the .gitmodules file is a good temporary fix for anyone that needs one though.

kito-cheng, 2021-09-09:
Hot fix: https://github.com/riscv-software-src/riscv-gnu-toolchain/pull/959

kito-cheng, 2021-09-09:
Let me know if the fix not work to you :)

jim-wilson, 2021-09-09:
Looks like something changed on the RISC-V International side, but I don't know exactly what changed.  When I clone riscv/riscv-gnu-toolchain now the "git remote -v" command says the remote is riscv/riscv-gnu-toolchain, and hence it works without Kito's latest patch.  When it was broken, git remote -v said the remote was riscv-software-src/riscv-gnu-toolchain.  Anyways, I don't know what changed and I don't know if we can rely on it, so I think we still need Kito's patch.  But at least with the current scheme old versions of riscv-gnu-toolchain should work.  If you clone riscv-software-src/riscv-gnu-toolchain then obviously you need Kito's patch.

kito-cheng, 2021-09-10:
I assume that PR has fix that issue, but feel free to reopen if anyone got this issue again.

closed 2021-09-10 by kito-cheng