riscv-collab/riscv-gnu-toolchain #961: Issue with installation command provided

franciscob1218, 2021-09-17:
I was not able to add programs that I needed and the terminal for ubuntu did not tell me it was not going to install it. I would add -y after the "install" so that it will grab "texinfo, bison, gpref" because they are found through GMU program mirrors.

Also, the copy for that command leaves a 01~ at the end as it will not pull that library, originally I thought that the  01 was a part of the install program.

jim-wilson, 2021-09-17:
It isn't clear what problems you are running into.

Are you talking about problems with the apt-get install command?  If you can't install some of the packages, then you won't be able to build the toolchain normally.

I don't understand the part about the 01~.  What command added that?

If you are having trouble building a toolchain, you may be able to install a pre-built one.  Some linux distros now have riscv64 cross compiler packages.  And there are pointers to a number of pre-built toolchains on the sifive.com/software web page.

TommyMurphyTM1234, 2022-09-27:
No update/activity for over a year and this doesn't seem to be an issue with the RISC-V GCC toolchain itself.

closed 2022-09-27 by TommyMurphyTM1234