riscv-collab/riscv-gnu-toolchain #962: RISC-V64 float/double Support

Kendidi, 2021-09-18:
Hi,

I have setup a riscv64 cross-compile environment and try to compile application that does floating point numbers arithmetic but encounter errors.

Toolchain was built with the following:

    ./configure --prefix=/opt/riscv64 --with-arch=rv64imafc --with-abi=lp64f

Then I tried to build code with the following:

    riscv64-unknown-elf-gcc -g -ffreestanding -O0 -Wno-int-to-pointer-cast  --specs=nosys.specs -Wl,--gc-sections -nostartfiles -mcmodel=medany -Wl,-T,riscv64-virt.ld  startup.s -o hello.elf main.c

If the code do not specify floating point values operation, it will compile just fine.  But when I do, I will encounter the following:

    /opt/riscv64/lib/gcc/riscv64-unknown-elf/10.2.0/libgcc.a(_clzsi2.o): in function `__clzdi2':
    /home/aie/riscv-gnu-toolchain/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libgcc/../../.././riscv-gcc/libgcc/libgcc2.c:710:(.text+0x12): relocation truncated to fit: R_RISCV_HI20 against symbol `__clz_tab' defined in .rodata section in /opt/riscv64/lib/gcc/riscv64-unknown-elf/10.2.0/libgcc.a(_clz.o)
    collect2: error: ld returned 1 exit status

Please advise on what I have done incorrectly?
And I wonder how can I switch soft-float and hard-float?
Thanks a lot in advance!

aswaterman, 2021-09-18:
The default code model (medium-low) only supports addresses in the lower 2 GiB.  I'm guessing your linker script uses addresses higher than that.

Try configuring the toolchain with `--with-cmodel=medany` and rebuild.

Kendidi, 2021-09-19:
Thank you @aswaterman!!  Yes, you are correct.  It can be built successfully now.

aswaterman, 2021-09-19:
Great!

closed 2021-09-19 by aswaterman