riscv-collab/riscv-gnu-toolchain #964: How to properly call the program constructors (C++ and special libraries)

useredsa, 2021-09-22:
Hello,

First of all, I must say I am going to pose a question that is not strictly related to this toolchain, but to gcc in general. I understand if you find the question out of topic and decide not to answer. If this is already answered in some online resource, I would gladly read it too. Thanks in advance.

Somehow, I ended up having to tweak the linker script and init functions for a new embedded riscv platform. I read some online resources and I have an implementation working. However, I would want to learn some things and receive some feedback. I cannot ask for it at my workplace because I don't know anybody that has done anything similar before.

My objective was to write a linker script and the necessary initialization that would come before the main function. And the following questions have arised:

1. Naming conventions: I started modyfing a linker script that had symbols like `__bss_start__` and `__bss_end__`. On the internet, however, others seem to prefer `__sbss` and `__ebss` for example.
1.1 Is there any convention I should follow for these names?
1.2 In general, we declare symbols with an underscore to avoid clashes with user symbols. However, should we use two underscores or just one? Or is it simply a matter of taste?
2. How can I be sure that I have declared all the important sections? Is there somewhere in the gcc internals documentation where all the sections produced by the compiler are listed? In particular, I'm also interested in the sections related to constructors.
3. Is this code correct to call all the constructors? See the bounds in the second loop. As far as I could test, It seems to work with simple programs. (You can check my linker script below, if you need it.)
```cpp
void _reset_handler() {
  // [...]
  int preinitSize = __preinit_array_end - __preinit_array_start;
  for (int i = 0; i < preinitSize; i++) {
    __preinit_array_start[i] ();
  }

  //TODO(useredsa) this could be used instead
  // but fails because `Init` is not found.
  // __libc_init_array();

  int initSize = __init_array_end - __init_array_start;
  // First pointer shall be ignored:
  // https://gcc.gnu.org/onlinedocs/gccint/Initialization.html.
  // NOTE(useredsa): Last position should be a zero, and also ignored,
  // but my experiments do not show that there is such an entry.
  for (int i = 1; i < initSize; i++) {
    __init_array_start[i] ();
  }
  // [...]
```
4. Why can't I call `__libc_init_array()` in the code above? `Init` is not defined, but should I define it?
5. What is inside the implementation of the `crtbegin.o` and `crtend.o`? I am using a pair provided by the toolchain but I don't know if I should write my own.

Linker script:
```
OUTPUT_ARCH(riscv)

/**
 * Last 8KB of RAM (0x00078000 - 0x0007FFFF) are left free and
 * can be used by the program to store possible results .
 */
_results_arr_size = DEFINED(_results_arr_size)? _results_arr_size : 0x8000;
_return_value_size = 0x4;

MEMORY
{
  IRAM (RX) : ORIGIN = 0x00000000 , LENGTH = 0x00040000
  DRAM (RWX) : ORIGIN = 0x00040000 , LENGTH = 0x00040000 - _return_value_size - _results_arr_size
}

/*
 * Reserve memory for heap and stack. 
 * The linker will issue an error if there is not enough memory.
 *
 * NOTE: The reserved heap and stack will be added to the bss column of the
 * binutils size command.
 */
_heap_size  = DEFINED(_heap_size)? _heap_size : 0;
_stack_size = DEFINED(_stack_size)? _stack_size : 4K;

/* program start point */
ENTRY(_start)

/**
 * The stack starts at the end of DRAM and grows downwards.
 * Full-descending stack; decrement first, then store.
 */
__stack = ORIGIN(DRAM) + LENGTH(DRAM);

SECTIONS
{
  .vectors MAX(0x00000000, ALIGN(4)) :
  {
    . = ALIGN(4);
    __vectors_start__ = .;

    *(.vectors*)

    . = ALIGN(4);
    __vectors_end__ = .;

  } > IRAM

  .text :
  {
    . = ALIGN(4);
    __text_start__ = .;

    *(.text*)

    . = ALIGN(4);
    __text_end__ = .;

  } > IRAM

  .rodata :
  {
    . = ALIGN(4);
    __rodata_start__ = .;

    *(.rodata*)

    . = ALIGN(4);
    __rodata_end__ = .;

  } > DRAM

  .data :
  {
    . = ALIGN(4);
    __data_start__ = .;

    *(.data*)

    . = ALIGN(4);
    __data_end__ = .;

  } > DRAM

  .bss :
  {
    . = ALIGN(4);
    __bss_start__ = .;

    *(.bss*)

    . = ALIGN(4);
    __bss_end__ = .;

  } > DRAM

  ._heap :
  {
    __heap_start__ = .;
    . = ALIGN(4);
    . = . + _heap_size;
    . = ALIGN(4);
    __heap_end__ = .;
  } > DRAM
  
  ._stack :
  {
    . = ALIGN(4);
    . = . + _stack_size;
    . = ALIGN(4);
  } > DRAM

}
```

My `crto.S` also for reference:

```
.section .text

	.global __rt_no_irq_handler

	__rt_no_irq_handler:
			j __rt_no_irq_handler

.section .vectors, "ax"

	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler
	j __rt_no_irq_handler

	.org 0x80
	.global _start

	_start:
		la sp, __stack
		j _reset_handler
```

Again, if you find anything that I am doing against convention, please tell me. Even better, if you have a reference to anything describing the conventions, I would be interested in reading it.

Best regards,

jim-wilson, 2021-09-23:
> 1. Naming conventions: I started modyfing a linker script that had symbols like `__bss_start__` and `__bss_end__`. On the internet, however, others seem to prefer `__sbss` and `__ebss` for example.
>    1.1 Is there any convention I should follow for these names?

Some of the names are historical.  _end probably goes back to Berkeley Unix, and maybe even the original AT&T Unix.  The ones that start and end with double underscores tend to be GNU extensions.  The important thing about these names is that various C libraries depend on them.  If they aren't there, or aren't in the right place, the C libraries may not work.  You can add your own symbols if you need them.

>    1.2 In general, we declare symbols with an underscore to avoid clashes with user symbols. However, should we use two underscores or just one? Or is it simply a matter of taste?

The ISO C standard says "All identifiers that begin with an underscore and either an uppercase letter or another underscore are always reserved for any use."  Hence these are safe for the toolchain to use.  Some historical names like _end violate this rule but can't easily be changed for historical reasons.  GNU folks prefer to use two underscores.

> 2. How can I be sure that I have declared all the important sections? Is there somewhere in the gcc internals documentation where all the sections produced by the compiler are listed? In particular, I'm also interested in the sections related to constructors.

The GNU linker has an option --orphan-handling=[place|warn|error|discard] where an orphan section is one that is not handled by the linker script.  So --orphan-handling=error will give an error if a section is missing from your linker script.

> 3. Is this code correct to call all the constructors? See the bounds in the second loop. As far as I could test, It seems to work with simple programs. (You can check my linker script below, if you need it.)

Shrug.  I suppose it looks reasonable.  I suggest you debug it if it doesn't work. 

> 1. Why can't I call `__libc_init_array()` in the code above? `Init` is not defined, but should I define it?

.init sections were deprecated about 20 years ago when the ELF standard added .init_array.  .init has the problem that you can't backtrace through it, and hence C++ EH doesn't work.  .init_array solves this problem.  It took maybe a decade for the tools to switch from .init to .init_array, but the tools still support .init because dropping it would break old code.  However, RISC-V is a new target, and has no old code that we need to be compatible with, so we did drop support for .init (and .fini) and have only .init_array (and .fini_array).  Since there is no .init section in RISC-V code, there will be no Init symbol that marks the start of the section.  There are a few other new ISAs that have also dropped support for .init sections.

Maybe you have old C library code that hasn't been ported to RISC-V, and still has references to the now obsolete .init section?

> 2. What is inside the implementation of the `crtbegin.o` and `crtend.o`? I am using a pair provided by the toolchain but I don't know if I should write my own.

Some support code for the compiler to make static constructors and destructors work, and to provide default constructors/destructors for some toolchain features like shared libraries and transactional memory.  If you are using C++ then you probably need some of the code in here.  If you aren't using C++, then you may not need any of it.

useredsa, 2021-09-23:
Hi @jim-wilson,

First of all, thanks for answering.

> Some of the names are historical. _end probably goes back to Berkeley Unix, and maybe even the original AT&T Unix. The ones that start and end with double underscores tend to be GNU extensions. The [...]

I understand that they may come from convention, but if they are not listed somewhere, how would I know which names to use? I mean, I could program things until something fails and fix it, but if I have to ship this so other people use it, I would like to be sure that it is always going to work.

My understanding is that this should be covered in the C library documentation, the only thing that would depend on them, right? Then I will try to find it there.

> The ISO C standard says "All identifiers that begin with an underscore and either an uppercase letter or another underscore are always reserved for any use." Hence these are safe for the toolchain [...]

Ok, so I should be safe using two underscores or one underscore and capital letter names if the C library is not using the same identifier (which would give an error anyway).

> The GNU linker has an option --orphan-handling=[place|warn|error|discard] where an orphan section is one that is not handled by the linker script. So --orphan-handling=error will give an error if a section is missing from your linker script.
> 
> >

I did not know about that option and indeed, it is very useful to include in tests. Thank you.
Still, my fear is to drop a section that did not appear during my tests. Although that seems difficult.

> Maybe you have old C library code that hasn't been ported to RISC-V, and still has references to the now obsolete .init section?

Well, it would not surprise me. I am forced to use the pulp toolchain v1, because it has support for some extra instructions for my processor. I am not the one to criticize because I am pretty much new to this, but I tend to find things like these (wrong library settings in this case).

In an ideal world, I would gladly provide a toolchain, with a correctly chosen C library and properly customized. Unfortunately, I cannot create a new toolchain, because even if I managed to do it properly, it would suppose a fairly big maintenance burden for the place I work for. I know I am not allowed to do it.

Given that I have to work with the toolchain we have, and that it also looks good to you, I will have to stick to this until something fails.

> Some support code for the compiler to make static constructors and destructors work, and to provide default constructors/destructors for some toolchain features like shared libraries and [...]

Yes, they are needed for C++ constructors (and for C functions marked as constructors too, actually). My question, if you happen to know the answer, would be «where do these files come from?». Are they provided by the C library? Are they crafted for a specific architecture? I guess... both?

Right now I am using these files from the pulp toolchain, but I don't know how they were generated. Now that I am in charge of the start up function, I would also like to know what do these files do (to read the source code I mean), and if I should write my own.

A little inconvenience is that our users need to compile with `-nostartfiles` manually and then link to this files,
which is error prone because one must come before anything else and the other one after.

----

Do you think using the same compiler binary but creating a new sysroot with its `crtbegin`, `crtend`, `crt0`, C library and linker script would help me solve these problems? As far as I understand, the users would no longer need to compile with `-nostartfiles`, `-T linker script`, link `crtbegin` and `crtend`, etc. And if the library was correct, I could even call `__libc_init_array`. Or is it not a good idea to create a different sysroot.

I just feel I am moving over a minefield of working software architecture for which I lack information.
Do you happen to know what could I start reading?

Thank you for your time and best regards,

jim-wilson, 2021-09-23:
> I understand that they may come from convention, but if they are not listed somewhere, how would I know which names to use? I mean, I could program things until something fails and fix it, but if I [...]

The GNU toolchain supports a lot of different C libraries, and each one has its own documentation.  So there is no master list.  And most C libraries probably don't document this.  Also some people have non-C-library code that uses some of these symbols.  You just have to accept that this is magic.  If you keep all of the symbols used in the GNU ld linker scripts, then you are OK.  If your code works, then you are OK.

> Yes, they are needed for C++ constructors (and for C functions marked as constructors too, actually). My question, if you happen to know the answer, would be «where do these files come from?». Are they provided by the C library? Are they crafted for a specific architecture? I guess... both?

crtbegin.o and crtend.o come from GCC.  They are built from the crtstuff.c file which has lots of ifdefs so that it can be built multiple ways to produce multiple object files.

> Do you think using the same compiler binary but creating a new sysroot with its `crtbegin`, `crtend`, `crt0`, C library and linker script would help me solve these problems? As far as I understand, [...]

GCC should be able to find its own crtbegin.o and crtend.o files.  Not clear what is wrong here.

> I just feel I am moving over a minefield of working software architecture for which I lack information.
> Do you happen to know what could I start reading?

I don't know if this kind of stuff is documented anywhere.

useredsa, 2021-09-24:
> The GNU toolchain supports a lot of different C libraries, and each one has its own documentation. So there is no master list. And most C libraries probably don't document this. Also some people [...]

I like the idea of defining the same as the GNU linker scripts. I will look for them and compare.
With that I think a big issue is solved.
Thank you very much.

> crtbegin.o and crtend.o come from GCC. They are built from the crtstuff.c file which has lots of ifdefs so that it can be built multiple ways to produce multiple object files.

Ok. I will check that file then. I guess they do not depend on the architecture then.

> GCC should be able to find its own crtbegin.o and crtend.o files. Not clear what is wrong here.
> 
> >

Indeed it can, but then it will also include its own `crt0.o`. Because I need to change the start up script, I am compiling with `-nostartfiles`, which also removes `crtbegin.o` and `crtend.o`. That is why I thought of creating a sysroot which would be a copy of the original but with a different linker script and `crt0.o` file.

> I don't know if this kind of stuff is documented anywhere.

Unfortunate I guess. Still, you were of a lot of help.

closed 2021-09-24 by useredsa