riscv-collab/riscv-gnu-toolchain #965: Can I compile a program with rv64i only?

wsnlance, 2021-09-24:
I am making a riscv simulator and it only supports for rv64i. So I hope the program only has instructions in rv64i. How can I make it?

TommyMurphyTM1234, 2021-09-24:
Pass `-march=rv64i` to the tools?

wsnlance, 2021-09-24:
@TommyMurphyTM1234 
Thanks for your reply but it does not work.
error: requested ABI requires '-march' to subsume the 'D' extension
I have read a similar article and it also talks about the available combinations of -march and -mabi.
Unfortunately there is no combination only include rv64i.
But there is a combination only include rv32i.
I will test it and comment later.

TommyMurphyTM1234, 2021-09-24:
What happens if you pass `-march=rv64i -mabi=lp64` to explicitly **not** select any floating point?

TommyMurphyTM1234, 2021-09-24:
> @TommyMurphyTM1234
> I have read a similar article and it also talks about the available combinations of -march and -mabi.
> Unfortunately there is no combination only include rv64i.

The article does not exhaustively enumerate all possible options.

wsnlance, 2021-09-24:
> What happens if you pass `-march=rv64i -mabi=lp64` to explicitly **not** select any floating point?
> I suspect that even with `-march=rv64i` the default ABI is effectively still `-mabi=lp64d`.
> 
> Where did you get the toolchain?
> Does it support multiple architecture options/extensions?

It shows another error and I can fix the error by changing -march=rv64i to -march=rv64imac, but that's not what I want.
I compile it by myself with code in this repository and I add --multilib when I compile it.

TommyMurphyTM1234, 2021-09-24:
> It shows another error 

What error?

> I compile it by myself with code in this repository and I add --multilib when I compile it.

What configure command do you use?
What multilibs to you configure it to build?

jim-wilson, 2021-09-24:
Are you building your own compiler or using one built by someone else?

If built by someone else, try running "riscv64-unknown-elf-gcc --print-multi-lib" which will tell you what arch/abi options are supported for linking.  -march will change the code generated by the compiler, but some libraries, like libc and libgcc are precompiled, and -march won't affect them.  So you may or may not have a precompiled version for the -march option you want to use.

If building your own compiler, you can pass --with-arch=rv64i --with-abi=lp64 when configuring the toolchain to get a compiler that supports exactly what you want.

If you don't want to build your own compiler, you can probably find one from someone else that does have the multilibs you want.  I think the SiFive one would work for instance.

There are far too many possible multilibs for us to build them all by default, so the default list has only a small subset, and rv64i/lp64 is not one of them.

TommyMurphyTM1234, 2021-09-27:
Any update or can you issue be closed @wsnlance ?

wsnlance, 2021-09-27:
I'm sorry for the delay but I'm not able to work on it in the weekend.
I successfully build a new toolchain with options --with-arch=rv64i --with-abi=lp64 and compile with -march=rv64i -mabi=lp64.

closed 2021-09-27 by wsnlance