riscv-collab/riscv-gnu-toolchain #966: Error occurred when building the toolchain

wsnlance, 2021-09-26:
I had built a toolchain before and I want to build a new one under the same environment.
Then the error occurred.
error: 'GATHER_STATISTICS' was not declared in this scope; did you mean 'GCC_STATISTICS'?
This error occurs in riscv-gcc/gcc/hash-map.h and riscv-gcc/gcc/hash-set.h

jim-wilson, 2021-09-26:
I would guess that there is something wrong with your environment.  Maybe you have compiler related environment variables set that are confusing the compiler.  Use "printenv" to check your environment for anything that shouldn't be there.

Another thing that can cause trouble is if you try to build a compiler using the exact same prefix you used before, but configure the compiler differently, or use different compiler sources.  Since we do installs during the build process, if there is already a compiler in prefix, this can confuse the build process and cause it to fail.  I don't think that it can cause this particular failure though.

Normally, the configure script should set GATHER_STATISTICS in the auto-host.h file in the gcc build directory.  That would be build-gcc-*-stage*/gcc/auto-host.h, depending on what kind of compiler you are trying to build, and which part of the build process it failed at.  Try looking at this file to see if it is OK, and has the GATHER_STATISTICS define.  If it is missing, try looking at the config.log file in the same dir for unexpected errors.  The configure script does feature tests to see what features are available, and some features won't be available on your host or target, so you need to look for something that should have worked but didn't.

TommyMurphyTM1234, 2022-09-27:
No activity/update for over a year and, as mentioned above, it seems to be an issue with the build environment rather than an issue with the RISC-V GCC toolchain or build process itself.

closed 2022-09-27 by TommyMurphyTM1234