riscv-collab/riscv-gnu-toolchain #967: Request username when updating submodule

Hssua, 2021-09-29:
After " git clone https://github.com/riscv/riscv-gnu-toolchain -b rvv-0.8.x ", I update the toolchain submodule by "git submodule update --init --recursive" . However, it requests username for updating riscv-glibc.

[image]

LevyHsu, 2021-09-29:
Check This:
https://github.blog/2020-12-15-token-authentication-requirements-for-git-operations/

“Beginning August 13, 2021, we will no longer accept account passwords when authenticating Git operations on GitHub.com.”

closed 2021-09-29 by Hssua

0x6770, 2021-09-30:
The url of `riscv-glibc` submodule in `rvv-0.8.x` branch is outdated, you can manually amend it by editing the `.gitmodules` file in the root to use `git://sourceware.org/git/glibc.git` as the master branch.