riscv-collab/riscv-gnu-toolchain #970: ld cannot find -lgcc_s

zaazad, 2021-10-04:
Hi, I'm trying to build [this project](https://github.com/microsoft/SEAL-Embedded) for risc-v target (both linux and elf toolchains). With linux toolchain, when I pass "-static" option to the compiler (-DCMAKE_C_FLAGS="-O3 -static -v"), I get the following error. But building without -static flag is successful.

/riscv64-unknown-linux-gnu/bin/ld: cannot find -lgcc_s

Also, in the case of elf toolchain, both w/ and w/o -static, it throws similar errors:
riscv64-unknown-elf/bin/ld: cannot find -lgcc_s

It seems that it is able to find -lgcc and -lgcc_eh but not -lgcc_s. I manually copied "libgcc_s.so" to the same folder as "libgcc.a" for linux toolchain, but still getting the same error. Also, I couldn't find libgcc_s.so in the elf toolchain lib folder. Any ideas what is going wrong? 

Thank you!
'''
riscv64-unknown-linux-gnu-gcc  --print-file-name=libgcc_s.so
/net/engnas/riscv/_install/bin/../lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/lib/../lib64/lp64d/libgcc_s.so

riscv64-unknown-linux-gnu-gcc  --print-file-name=libgcc.a
/net/engnas/riscv/_install/bin/../lib/gcc/riscv64-unknown-linux-gnu/11.1.0/libgcc.a

/net/engnas/riscv_elf_toolchain/_install/bin/riscv64-unknown-elf-gcc --print-file-name=libgcc.a   
/net/engnas/riscv_elf_toolchain/_install/bin/../lib/gcc/riscv64-unknown-elf/11.1.0/libgcc.a

 /net/engnas/riscv_elf_toolchain/_install/bin/riscv64-unknown-elf-gcc --print-file-name=libgcc_s.so
libgcc_s.so
'''
Here are the logs of different builds.
Linux toolchain with -static flag:
'''
Linking C executable bin/seal_embedded_tests
Using built-in specs.
COLLECT_GCC=/net/engnas/riscv/_install/bin/riscv64-unknown-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/net/engnas/riscv/_install/bin/../libexec/gcc/riscv64-unknown-linux-gnu/11.1.0/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: /net/engnas/riscv/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-linux-gnu --prefix=/net/engnas/riscv/riscv-gnu-toolchain/../_install --with-sysroot=/net/engnas/riscv/riscv-gnu-toolchain/../_install/sysroot --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=.././riscv-gcc --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket 'CFLAGS_FOR_TARGET=-O2  -mno-relax -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2  -mno-relax -mcmodel=medlow'
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 11.1.0 (GCC) 
COMPILER_PATH=/net/engnas/riscv/_install/bin/../libexec/gcc/riscv64-unknown-linux-gnu/11.1.0/:/net/engnas/riscv/_install/bin/../libexec/gcc/:/net/engnas/riscv/_install/bin/../lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/bin/
LIBRARY_PATH=/net/engnas/riscv/_install/bin/../lib/gcc/riscv64-unknown-linux-gnu/11.1.0/:/net/engnas/riscv/_install/bin/../lib/gcc/:/net/engnas/riscv/_install/bin/../lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/lib/../lib64/lp64d/:/net/engnas/riscv/_install/bin/../sysroot/lib/../lib64/lp64d/:/net/engnas/riscv/_install/bin/../lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/lib/:/net/engnas/riscv/_install/bin/../sysroot/lib64/lp64d/:/net/engnas/riscv/_install/bin/../sysroot/usr/lib64/lp64d/:/net/engnas/riscv/_install/bin/../sysroot/lib/
COLLECT_GCC_OPTIONS='-O3' '-static' '-v' '-g' '-o' 'bin/seal_embedded_tests' '-mtune=rocket' '-march=rv64imafdc' '-mabi=lp64d' '-march=rv64imafdc' '-dumpdir' 'bin/seal_embedded_tests.'
 /net/engnas/riscv/_install/bin/../libexec/gcc/riscv64-unknown-linux-gnu/11.1.0/collect2 -plugin /net/engnas/riscv/_install/bin/../libexec/gcc/riscv64-unknown-linux-gnu/11.1.0/liblto_plugin.so -plugin-opt=/net/engnas/riscv/_install/bin/../libexec/gcc/riscv64-unknown-linux-gnu/11.1.0/lto-wrapper -plugin-opt=-fresolution=/scratch/ccVTHSK6.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_eh -plugin-opt=-pass-through=-lc --sysroot=/net/engnas/riscv/_install/bin/../sysroot -melf64lriscv -static -o bin/seal_embedded_tests /net/engnas/riscv/_install/bin/../sysroot/usr/lib64/lp64d/crt1.o /net/engnas/riscv/_install/bin/../lib/gcc/riscv64-unknown-linux-gnu/11.1.0/crti.o /net/engnas/riscv/_install/bin/../lib/gcc/riscv64-unknown-linux-gnu/11.1.0/crtbeginT.o -L/net/engnas/riscv/_install/bin/../lib/gcc/riscv64-unknown-linux-gnu/11.1.0 -L/net/engnas/riscv/_install/bin/../lib/gcc -L/net/engnas/riscv/_install/bin/../lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/lib/../lib64/lp64d -L/net/engnas/riscv/_install/bin/../sysroot/lib/../lib64/lp64d -L/net/engnas/riscv/_install/bin/../lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/lib -L/net/engnas/riscv/_install/bin/../sysroot/lib64/lp64d -L/net/engnas/riscv/_install/bin/../sysroot/usr/lib64/lp64d -L/net/engnas/riscv/_install/bin/../sysroot/lib CMakeFiles/seal_embedded_tests.dir/test/fft_tests.c.o CMakeFiles/seal_embedded_tests.dir/test/modulo_tests.c.o CMakeFiles/seal_embedded_tests.dir/test/network_tests.c.o CMakeFiles/seal_embedded_tests.dir/test/sample_tests.c.o CMakeFiles/seal_embedded_tests.dir/test/uintmodarith_tests.c.o CMakeFiles/seal_embedded_tests.dir/test/uintops_tests.c.o CMakeFiles/seal_embedded_tests.dir/test/ntt_tests.c.o CMakeFiles/seal_embedded_tests.dir/test/ckks_tests_encode.c.o CMakeFiles/seal_embedded_tests.dir/test/ckks_tests_common.c.o CMakeFiles/seal_embedded_tests.dir/test/ckks_tests_asym.c.o CMakeFiles/seal_embedded_tests.dir/test/ckks_tests_sym.c.o CMakeFiles/seal_embedded_tests.dir/test/api_tests.c.o CMakeFiles/seal_embedded_tests.dir/test/main.c.o -lm -lgcc_s -lc bin/libseal_embedded.a -lm --start-group -lgcc -lgcc_eh -lc --end-group /net/engnas/riscv/_install/bin/../lib/gcc/riscv64-unknown-linux-gnu/11.1.0/crtend.o /net/engnas/riscv/_install/bin/../lib/gcc/riscv64-unknown-linux-gnu/11.1.0/crtn.o
**/net/engnas/riscv/_install/bin/../lib/gcc/riscv64-unknown-linux-gnu/11.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find -lgcc_s**
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/seal_embedded_tests.dir/build.make:284: bin/seal_embedded_tests] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:177: CMakeFiles/seal_embedded_tests.dir/all] Error 2
'''

Elf toolchain with -static flag:
'''
[100%] Linking C executable bin/seal_embedded_tests
Using built-in specs.
COLLECT_GCC=/net/engnas/riscv_elf_toolchain/_install/bin/riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/net/engnas/riscv_elf_toolchain/_install/bin/../libexec/gcc/riscv64-unknown-elf/11.1.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /net/engnas/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/net/engnas/riscv-gnu-toolchain/../_install --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-sysroot=/net/engnas/riscv-gnu-toolchain/../_install/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././riscv-gcc --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket 'CFLAGS_FOR_TARGET=-Os   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os   -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 11.1.0 (GCC) 
COMPILER_PATH=/net/engnas/riscv_elf_toolchain/_install/bin/../libexec/gcc/riscv64-unknown-elf/11.1.0/:/net/engnas/riscv_elf_toolchain/_install/bin/../libexec/gcc/:/net/engnas/riscv_elf/riscv_elf_toolchain/_install/bin/../lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/bin/
LIBRARY_PATH=/net/engnas/riscv_elf_toolchain/_install/bin/../lib/gcc/riscv64-unknown-elf/11.1.0/:/net/engnas/riscv_elf_toolchain/_install/bin/../lib/gcc/:/net/engnas/riscv_elf_toolchain/_install/bin/../lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/lib/:/net/engnas/riscv_elf_toolchain/_install/bin/../riscv64-unknown-elf/lib/
COLLECT_GCC_OPTIONS='-O3' '-static' '-v' '-g' '-o' 'bin/seal_embedded_tests' '-mtune=rocket' '-march=rv64imafdc' '-mabi=lp64d' '-march=rv64imafdc' '-dumpdir' 'bin/seal_embedded_tests.'
[... cut ...] riscv_elf_toolchain/_install/bin/../lib/gcc/riscv64-unknown-elf/11.1.0/crtend.o
**/net/engnas/riscv_elf_toolchain/_install/bin/../lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/bin/ld: cannot find -lgcc_s**
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/seal_embedded_tests.dir/build.make:284: bin/seal_embedded_tests] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:177: CMakeFiles/seal_embedded_tests.dir/all] Error 2
gmake: *** [Makefile:103: all] Error 2
'''

TommyMurphyTM1234, 2021-10-04:
Any use (not RISC-V specific)?

https://stackoverflow.com/questions/51998912/when-statically-linking-a-library-getting-linker-error-cannot-find-lgcc-s

TommyMurphyTM1234, 2021-10-04:
What if you add `-static-libgcc`?

jim-wilson, 2021-10-04:
As the reference Tommy Murphy pointed to explains, when using --static, you shouldn't be trying to link with -lgcc_s as there is no static version of this.  So I would suspect a bug in the package you are trying to compile.  Looking at the cmake support files, I found in the file device/CMakeLists.txt the line
        target_link_libraries(seal_embedded_bench PRIVATE m gcc_s c seal_embedded)
which is apparently trying to link unconditionally with -lgcc_s which is wrong when --static is used.

jim-wilson, 2021-10-04:
The elf toolchain doesn't support shared libraries, so it will fail with or without --static, as the package is always trying to link with libgcc_s.  The linux toolchain will only work without --static.

TommyMurphyTM1234, 2021-10-04:
> As the reference Tommy Murphy pointed to explains

Apologies, I originally posted this but then deleted it as I wasn't sure if it was relevant after all...

https://stackoverflow.com/questions/51998912/when-statically-linking-a-library-getting-linker-error-cannot-find-lgcc-s

Posting it again so that @jim-wilson's comment makes sense.

zaazad, 2021-10-05:
Thank you Jim and Tommy! 
I modified the cmake files to link gcc when gcc_s doesn't exist, and it works fine now.
I'll open an issue in SEAL repo and report this.

closed 2021-10-05 by zaazad