riscv-collab/riscv-gnu-toolchain #987: How to compile GCC only and not G++

uniquepointer, 2021-11-01:
I only really need `gcc` and not `g++` so i was wondering if there was a way with the provided `configure` script that was provided to build `gcc` only that I might've missed?!?

uniquepointer, 2021-11-01:
I just replaced every single line like the following with c only
https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/Makefile.in#L623
I'm still compiling but it seems like it worked

kito-cheng, 2021-11-01:
Yeah, you found the right way :)

uniquepointer, 2021-11-01:
Yeah :D however it would be nice to add an option to the configure script!

TommyMurphyTM1234, 2021-11-01:
> Yeah :D however it would be nice to add an option to the configure script!

If I recall correctly (can't check for sure right now) [crosstool-NG](https://crosstool-ng.github.io/)  has an option to enable or disable the building of the g++ C++ tools so maybe it leverages something already in the GCC configure script?

jim-wilson, 2021-11-02:
That is exactly what the original poster is using.  We just don't expose it in the riscv-gnu-toolchain toplevel configure and Makefile.

TommyMurphyTM1234, 2021-11-02:
> That is exactly what the original poster is using. 

They gave no indication that they're using crosstool-NG.

uniquepointer, 2021-11-06:
So is it ok to close this? Or should I keep it open if there is plans for adding an option to this repo?

jim-wilson, 2021-11-07:
They are using riscv-gnu-toolchain.  They did use the gcc --enable-language option.  There just isn't any way to use that from the toplevel configure.  So they had to modify the Makefile.

Someone would have to do some work to make this available from the toplevel configure.  That is the only problem here.

TommyMurphyTM1234, 2021-11-08:
So, does anybody know if/why riscv-gnu-toolchain needs a custom configure script and cannot just use the upstream/master gcc configure script as is? Particularly given that crosstool-NG presumably uses the latter when building the RISC-V toolchain.

jim-wilson, 2021-11-08:
Because our configure script does riscv-gnu-toolchain stuff.  The gcc configure script only handles gcc, it doesn't handle any of the other interesting things that the riscv-gnu-toolchain configure script does.  This is the same reason crosstools-ng has their own configure script on top of the gcc configure script, well, actually they use a linux kernel style configury, but it is the same issue.  You need toolchain configuration stuff that the gcc configure script doesn't do.  And then you call the gcc configure script from the toplevel configure script.

For enable-languages, there is the issue that we have different defaults for different targets.  The newlib toolchain only builds C and C++.  The glibc Linux toolchain builds C, C++, and Fortran.  There is also a musl Linux toolchain that builds only C and C++.  So we can't just pass it through unless someone does some work to add minimal support to provide the defaults if it wasn't specified on the command line.

ajf58, 2021-12-29:
It looks like #1002 means that glibc no longer build for C++.

jim-wilson, 2021-12-29:
#1002 is a build script problem, and has nothing to do with whether glibc works for C++.  It still does.

ajf58, 2022-01-01:
Thanks for the correction/clarification, @jim-wilson .

cmuellner, 2022-09-29:
C, C++ and Fortran are needed for many benchmark suites, therefore I consider the selection a reasonable one.
If we start arguing to remove C++, others might argue to add Ada, D, Go and ObjC.
What could be done is to configure the language selection with our configure script (and forward it to the build step).

@TommyMurphyTM1234: not all occurrences need to change, only the stage2 compiler and the native compiler.

With that said (and the hint how to remove the C++ support by changing Makefile.in) I'm closing this issue.
If someone wants to implement a configurable list of language support, go ahead and prepare a PR.

closed 2022-09-29 by cmuellner