riscv-collab/riscv-gnu-toolchain #989: Decimal to Floating Point

amithmath, 2021-11-09:
Hello Jim and All,

I am wondering how decimals are converted to floats and floats are converted back to decimal in riscv-gcc for hard FPU? Consider the following example:

#include<stdio.h>
#include<stdlib.h>

int main()
{

float a,b,c;

a= ((float)rand()/(float)(RAND_MAX));
b= ((float)rand()/(float)(RAND_MAX));

c = a+b;

return 0;

}

once we assign a and b from rand(), the random numbers like: 0.439543994, 10449.5503004, etc. converted to floats from strtod in riscv-newlib/newlib/libc/stdlib/strtod.c and converted back to decimal from riscv-newlib/newlib/libc/stdlib/dtoa.c? Or any other functions?

I think the conversion functions (real_from_string and real_to_decimal ) in https://github.com/riscv-collab/riscv-gcc/blob/5964b5cd72721186ea2195a7be8d40cfe6554023/gcc/real.c are for soft FPU.

Thanks,
-Amit

TommyMurphyTM1234, 2021-11-09:
Have you looked at the disssembly/list file for your compiled and linked program?

amithmath, 2021-11-09:
Yes, I have configured riscv-gcc compiler for hard float instructions and built it.  I disassembled the code  but there is no info. about conversion function.  Following is the disassemble code:

  .file   "f_test.c"
        .option nopic
        .attribute arch, "rv32i2p0_m2p0_a2p0_f2p0"
        .attribute unaligned_access, 0
        .attribute stack_align, 16
        .text
        .align  2
        .globl  main
        .type   main, @function
main:
        addi    sp,sp,-32
        sw      ra,28(sp)
        sw      s0,24(sp)
        addi    s0,sp,32
        call    rand
        mv      a5,a0
        fcvt.s.w        fa4,a5
        lui     a5,%hi(.LC0)
        flw     fa5,%lo(.LC0)(a5)
        fdiv.s  fa5,fa4,fa5
        fsw     fa5,-20(s0)
        call    rand
        mv      a5,a0
        fcvt.s.w        fa4,a5
        lui     a5,%hi(.LC0)
        flw     fa5,%lo(.LC0)(a5)
        fdiv.s  fa5,fa4,fa5
        fsw     fa5,-24(s0)
        flw     fa4,-20(s0)
        flw     fa5,-24(s0)
        fadd.s  fa5,fa4,fa5
        fsw     fa5,-28(s0)
		li      a5,0
        mv      a0,a5
        lw      ra,28(sp)
        lw      s0,24(sp)
        addi    sp,sp,32
        jr      ra
        .size   main, .-main
        .section        .rodata
        .align  2
.LC0:
        .word   2142240768
        .ident  "GCC: (GNU) 9.2.0"

I don't think in the disassembled code one can see conversion functions because conversion functions are not part of RISCV ISA. In my view, before converting to target machine code, the conversion functions are carried out in front end of the compiler. Perhaps I may be wrong...

TommyMurphyTM1234, 2021-11-09:
Aren't the `fcvt` instructions doing the conversion from integer to float?

amithmath, 2021-11-09:
In my view,  fcvt instruction is not doing conversion from integer to human readable decimal format. I think this instruction does convert to integer to float, when I ask the above code to print 'a' like:

#include<stdio.h>
#include<stdlib.h>

int main()
{

float a,b,c;

a= ((float)rand()/(float)(RAND_MAX));
b= ((float)rand()/(float)(RAND_MAX));
c = a+b;

printf("%f\n",a);

return 0;

}

disassemble code look like:

.file   "f_test.c"
        .option nopic
        .attribute arch, "rv32i2p0_m2p0_a2p0_f2p0"
        .attribute unaligned_access, 0
        .attribute stack_align, 16
        .text
        .globl  __extendsfdf2
        .section        .rodata
        .align  2
.LC1:
        .string "%f\n"
        .text
        .align  2
        .globl  main
        .type   main, @function
main:
        addi    sp,sp,-32
        sw      ra,28(sp)
        sw      s0,24(sp)
        addi    s0,sp,32
        call    rand
        mv      a5,a0
        fcvt.s.w        fa4,a5
        lui     a5,%hi(.LC0)
        flw     fa5,%lo(.LC0)(a5)
        fdiv.s  fa5,fa4,fa5
        fsw     fa5,-20(s0)
        call    rand
        mv      a5,a0
        fcvt.s.w        fa4,a5
        lui     a5,%hi(.LC0)
        flw     fa5,%lo(.LC0)(a5)
       fdiv.s  fa5,fa4,fa5
        fsw     fa5,-24(s0)
        flw     fa4,-20(s0)
        flw     fa5,-24(s0)
        fadd.s  fa5,fa4,fa5
        fsw     fa5,-28(s0)
        flw     fa0,-20(s0)
        call    __extendsfdf2
        mv      a4,a0
        mv      a5,a1
        mv      a2,a4
        mv      a3,a5
        lui     a5,%hi(.LC1)
        addi    a0,a5,%lo(.LC1)
        call    printf
        li      a5,0
        mv      a0,a5
        lw      ra,28(sp)
        lw      s0,24(sp)
        addi    sp,sp,32
        jr      ra
        .size   main, .-main
        .section        .rodata
        .align  2
.LC0:
        .word   2142240768
        .ident  "GCC: (GNU) 9.2.0"

It will first call __extendsfdf2 to extend float to double and then call printf. If I type: find -iname printf* I got these two file:

riscv-newlib/newlib/libc/stdio/printf.c
riscv-glibc/stdio-common/printf.c

But there is no information/logic about decimal conversion.

TommyMurphyTM1234, 2021-11-09:
> In my view, fcvt instruction is not doing conversion from integer to human readable decimal format 

No, it's not. But that's not what you originally asked about, it doesn't relate to the subject of the issue, and your original code did not have any call to `printf()`. If you want to see what's happening when you call ` printf ()` to output a floating point value then just look at the disassembly/list file for the program and you'll see exactly what happens.

Maybe it would help if you explained in a bit more detail what exactly you're looking for here and what's causing you confusion?

amithmath, 2021-11-09:
I have custom float point unit (FPU) which has 1 sign bit, 11 exponent bits and 20 mantissa bits, the thing is, I have to pass  random numbers to custom FPU and perform custom FPU operations such as addition, multiplication, division, etc. obtain results from custom FPU and print it.  I am wondering how to pass custom FP values and display it?

jim-wilson, 2021-11-09:
The code in real.c is used at compile time.  For a cross compiler, the host and target may not have the exact same FP format, so we can't expect to do FP arithmetic natively in the compiler.  We emulate it via the code in real.c.  There are also other issues, e.g. the code being compiled might have specified its own rounding mode different from what the compiler is using.  The code being compiled might be creating a signalling nan, and we don't want to create a signal in the compiler while compiling the code.  Etc.  It is just easier if we emulate all of the FP arithmetic in the compiler.

At run time, all of the conversions will happen via FP instructions or library calls.

amithmath, 2021-11-09:
In my view, one has to modify custom library functions for exceptions, overflow and underflow issues, which I will modify later perhaps I may be wrong since I am not compiler design expert. But at first I am interested how to pass generated random numbers from above program to custom hard FPU and print the result. 

Jim: may I ask how at run rime conversions are performed? What are the conversion functions or library calls for hard FPU?

jim-wilson, 2021-11-09:
Run time conversions depend on the target.  If the target has an instruction, and the instruction does the right thing, then the instruction will be used.  Otherwise, a function call will be used, which could be a call to a target dependent library (e.g. ARM has a few of these) or it could be a generic function like newlib dtoa or scanf.

amithmath, 2021-11-09:
Thanks Jim. I think scanf for reading FP data from console. How about converting decimal to float from generated random numbers? Is the riscv-newlib/newlib/libc/stdlib/strtod.c used from string decimal to IEEE-754 float?

jim-wilson, 2021-11-09:
rand() returns an int which you cast to float.  So this will be a fp conversion instruction for a hard FP target, or a call to target specific function for some soft-float targets, or a call to a libgcc function e.g. __floatsidf.  The code for __floatsidf can be found in libgcc/fp-bit.[ch] which gets configured depending on the target's needs.

amithmath, 2021-11-10:
Got it Jim.  May I ask how about floats to decimal routines for hard FPU, I mean to display results? Where are the routines located?

jim-wilson, 2021-11-10:
If you are printing a fp value as a string, it works the same on soft-float or hard float.  You call a function to convert the FP value to a decimal string.  In newlib, this is newlib/libc/stdlib/dtoa.c.

closed 2021-11-10 by amithmath