riscv-collab/riscv-gnu-toolchain #993: fp8 and fp16 instructions

amithmath, 2021-11-19:
Hello Jim and All,

I am wondering, how to add fp8 and fp16 data type and custom instructions to RISC-V32 elf?  I think one has to add fp8 and fp16 in the file: https://github.com/riscv-collab/riscv-gcc/blob/5964b5cd72721186ea2195a7be8d40cfe6554023/gcc/c-family/c-common.c after line 482 but I am not getting how to generate instructions for hard FPU? 

Thanks,
-Amit

TommyMurphyTM1234, 2021-11-19:
Any use (on the general question of adding custom [RISC-V] instructions to GCC/binutils)?

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/837
- https://hsandid.github.io/posts/risc-v-custom-instruction/
- https://forums.sifive.com/t/adding-custom-instruction-to-riscv-toolchain/4044

amithmath, 2021-11-19:
Thanks Tommy. Let me study all links.

jim-wilson, 2021-11-19:
On Thu, Nov 18, 2021 at 11:39 PM amithmath ***@***.***> wrote:

> I am wondering, how to add fp8 and fp16 data type and custom instructions
> to RISC-V32 elf? I think one has to add fp8 and fp16 in the file: [...]

There is fp16 (IEEE binary16) support in riscv-gnu-toolchain on the
rvv-integration branch.  I expect this will be upstreamed when the zfh
extension gets ratified, but may not make it into the next gcc release.

The first step is to define an HFmode in the riscv-modes.def file.
riscv-modes.def:FLOAT_MODE (HF, 2, 0);
riscv-modes.def:ADJUST_FLOAT_FORMAT (HF, &ieee_half_format);

The second step is to add a type, e.g. __fp16
  /* _Float16 is C specific.  So we need a language independent type for
     half floats.  Use __fp16 same as the arm/aarch64 ports.  */
  fp16_type_node = make_node (REAL_TYPE);
  TYPE_PRECISION (fp16_type_node) = 16;
  layout_type (fp16_type_node);
  (*lang_hooks.types.register_builtin_type) (fp16_type_node, "__fp16");

The third step is to add HFmode patterns to the riscv.md file.  Search the
md file for references to HF.

The fourth step would be added soft-float routines in libgcc, which is
mainly a configuration change.

You might also want to add an option to turn it on/off.  This support
should really only be on if a user asks to turn it on.  Plus you should
depend on a -march flag as to whether we can do HFmode in hardware or not.
This involves changes to file in common/config/riscv.

You can try looking at how other ports implement HFmode support.  I don't
know if there is any QFmode support in gcc.  If you want bfloat16 instead
of binary16 then I don't know offhand if gcc has support for that.

Jim

amithmath, 2021-11-23:
Thanks Jim.

TommyMurphyTM1234, 2022-09-29:
No update/activity in almost a year. Anybody know if this is still an issue, if it can be closed, or if it needs to remain open as an enhancement issue to add further FP precision support?

amithmath, 2024-08-18:
is risc-v zfh extension ratified?

cmuellner, 2024-08-18:
> is risc-v zfh extension ratified?

Zfh was ratified in November 2021:
  https://wiki.riscv.org/display/HOME/Ratified+Extensions

amithmath, 2024-08-18:
Thanks. Does risc-v gcc,g++ and gfortran compilers support fp8/fp16?  If yes, please point me which risc-v gnu tool chain

TommyMurphyTM1234, 2024-08-18:
`Zfh` support was added in GCC 13 (can't determine which minor version added it though because of the way that GCC only publishes consolidated release notes/change logs covering all minor versions of a major version):

- https://gcc.gnu.org/gcc-13/changes.html

`riscv-gnu-toolchain` still uses GCC 13.2:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1341

The update to GCC 14 is still pending: 

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1531

If `riscv-gnu-toolchain` with GCC 13.2 doesn't have it then try building the toolchain with later out of tree GCC sources from upstream using `configure ... --with-gcc-src=...` . You may also need a later Binutils, standard library, etc.

TommyMurphyTM1234, 2024-08-18:
> `Zfh` support was added in GCC 13 (can't determine which minor version added it though because of the way that GCC only publishes consolidated release notes/change logs covering all minor versions of a major version):
> 
> * https://gcc.gnu.org/gcc-13/changes.html

Curiously I can't find any mention of `Zfh` in any of the 13.x minor release change logs:

- https://gcc.gnu.org/gcc-13/
- https://github.com/gcc-mirror/gcc/blob/releases/gcc-13.1.0/ChangeLog
- https://github.com/gcc-mirror/gcc/blob/releases/gcc-13.2.0/ChangeLog
- https://github.com/gcc-mirror/gcc/blob/releases/gcc-13.3.0/ChangeLog

TommyMurphyTM1234, 2024-08-18:
The latest CI build (using GCC 13.2.0 + Binutils 2.43):

- https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2024.08.06

doesn't complain when `-march` includes the `zfh` extension but it does fails when `Fp16` is used as in this post:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/993#issuecomment-1478356021
```
wget https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2024.08.06/riscv64-elf-ubuntu-22.04-gcc-nightly-2024.08.06-nightly.tar.gz
tar xzvf https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2024.08.06/riscv64-elf-ubuntu-22.04-gcc-nightly-2024.08.06-nightly.tar.gz
cd riscv/bin

cat test.c
void f(Fp16 x);

./riscv64-unknown-elf-gcc -march=rv64gc_zfh test.c -c
test.c:1:8: error: unknown type name 'Fp16'
    1 | void f(Fp16 x);
      |        ^~~~
```

TommyMurphyTM1234, 2024-08-18:
> The latest CI build (using GCC 13.2.0 + Binutils 2.43):
> 
> * https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2024.08.06
> 
> doesn't complain when `-march` includes the `zfh` extension but it does fails when `Fp16` is used as in this post:
> 
> * [fp8 and fp16 instructions #993 (comment)](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/993#issuecomment-1478356021)
> 
> ```
> wget https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2024.08.06/riscv64-elf-ubuntu-22.04-gcc-nightly-2024.08.06-nightly.tar.gz
> tar xzvf https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2024.08.06/riscv64-elf-ubuntu-22.04-gcc-nightly-2024.08.06-nightly.tar.gz
> cd riscv/bin
> 
> cat test.c
> void f(Fp16 x);
> 
> ./riscv64-unknown-elf-gcc -march=rv64gc_zfh test.c -c
> test.c:1:8: error: unknown type name 'Fp16'
>     1 | void f(Fp16 x);
>       |        ^~~~
> ```
> 
> (Same with the Linux rather than the bare-metal/Newlib variant of that toolchain).

Oh - looks like `_Float16` is the correct type to use.
```
cat test.c
void f(_Float16 x);

./riscv64-unknown-linux-gnu-gcc  -march=rv64gc_zfh test.c -c

file test.o
test.o: ELF 64-bit LSB relocatable, UCB RISC-V, RVC, double-float ABI, version 1 (SYSV), not stripped
```
`_Float8` doesn't work though - and maybe it's not supposed to?
```
cat test.c
void f(_Float8 x);

./riscv64-unknown-linux-gnu-gcc  -march=rv64gc_zfh test.c -c
test.c:1:8: error: unknown type name '_Float8'; did you mean '_Float128'?
    1 | void f(_Float8 x);
      |        ^~~~~~~
      |        _Float128
```

```

amithmath, 2024-08-18:
I am trying to build riscv-gnu-tool-chain with this repo: https://github.com/riscv-collab/riscv-gnu-toolchain to test __Float16 
./configure --prefix=/opt/riscv --disable-linux  --with-arch=rv32gc --with-abi=ilp32

make -j$(nproc) all  but I am getting the error: cc/gcc/config/riscv/genrvv-type-indexer.cc:118:30: error: ‘log2’ is not a member of ‘std and gcc/gcc/config/riscv/genrvv-type-indexer.cc:120:30: error: ‘log2’ is not a member of ‘std’

TommyMurphyTM1234, 2024-08-18:
Why don't you just use the pre-built CI toolchain binaries since they already seem to support `Zfh`?

> I am trying to build riscv-gnu-tool-chain with this repo: https://github.com/riscv-collab/riscv-gnu-toolchain to test __Float16 ./configure --prefix=/opt/riscv --disable-linux --with-arch=rv32gc --with-abi=ilp32

Why `--with-abi=ilp32` and not, say, `--with-abi=ilp32d` which arguably better matches the default architecture of `rv32gc` aka `rv32imafdc`?

> make -j$(nproc) all but I am getting the error: cc/gcc/config/riscv/genrvv-type-indexer.cc:118:30: error: ‘log2’ is not a member of ‘std and gcc/gcc/config/riscv/genrvv-type-indexer.cc:120:30: error: ‘log2’ is not a member of ‘std’

Details of your `clone`. `configure` and `make` commands and a build log?
Better to do a non parallel build without any `-j$(nproc)` option when tryng to investigate build problems.

amithmath, 2024-08-18:
I guess you meant pre-built CI toolchain binaries from:
wget https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2024.08.06/riscv64-elf-ubuntu-22.04-gcc-nightly-2024.08.06-nightly.tar.gz
tar xzvf https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2024.08.06/riscv64-elf-ubuntu-22.04-gcc-nightly-2024.08.06-nightly.tar.gz
cd riscv/bin ?
I did try riscv/bin/riscv64-unknown-gcc  -march=rv64gc_zfh test.c -c
I am getting following error: 
/lib64/libc.so.6: version `GLIBC_2.32' not found (required by riscv/bin/riscv64-unknown-elf-gcc)

If not, please point me  pre-built CI toolchain binaries

TommyMurphyTM1234, 2024-08-18:
I already linked to the CI builds above:

- https://github.com/riscv-collab/riscv-gnu-toolchain/releases

> I am getting following error:
> /lib64/libc.so.6: version `GLIBC_2.32' not found (required by riscv/bin/riscv64-unknown-elf-gcc)

You're probably using the wrong one for your Linux/GLIBC version.

TommyMurphyTM1234, 2024-08-18:
To find the version of GLIBC that your Linux distro uses:
```
ldd --version
ldd (Ubuntu GLIBC 2.35-0ubuntu3.8) 2.35
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
```
The CI builds are for Ubuntu 20.04 and 22.04 but should work on other distros with a GLIBC greater than or equal to the one that the toolchain was built to expect:
```
wget https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2024.08.06/riscv64-elf-ubuntu-20.04-gcc-nightly-2024.08.06-nightly.tar.gz
tar xzvf riscv64-elf-ubuntu-20.04-gcc-nightly-2024.08.06-nightly.tar.gz
mv riscv riscv-20.04

ldd -v riscv-20.04/bin/riscv64-unknown-elf-gcc
        linux-vdso.so.1 (0x00007ffcc4df5000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f5f5c50b000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f5f5c73f000)

        Version information:
        riscv-20.04/bin/riscv64-unknown-elf-gcc:
                ld-linux-x86-64.so.2 (GLIBC_2.3) => /lib64/ld-linux-x86-64.so.2
                libc.so.6 (GLIBC_2.9) => /lib/x86_64-linux-gnu/libc.so.6
                libc.so.6 (GLIBC_2.7) => /lib/x86_64-linux-gnu/libc.so.6
                libc.so.6 (GLIBC_2.14) => /lib/x86_64-linux-gnu/libc.so.6
                libc.so.6 (GLIBC_2.3) => /lib/x86_64-linux-gnu/libc.so.6
                libc.so.6 (GLIBC_2.4) => /lib/x86_64-linux-gnu/libc.so.6
                libc.so.6 (GLIBC_2.11) => /lib/x86_64-linux-gnu/libc.so.6
                libc.so.6 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libc.so.6
                libc.so.6 (GLIBC_2.3.4) => /lib/x86_64-linux-gnu/libc.so.6
        /lib/x86_64-linux-gnu/libc.so.6:
                ld-linux-x86-64.so.2 (GLIBC_2.2.5) => /lib64/ld-linux-x86-64.so.2
                ld-linux-x86-64.so.2 (GLIBC_2.3) => /lib64/ld-linux-x86-64.so.2
                ld-linux-x86-64.so.2 (GLIBC_PRIVATE) => /lib64/ld-linux-x86-64.so.2

wget https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2024.08.06/riscv64-elf-ubuntu-22.04-gcc-nightly-2024.08.06-nightly.tar.gz
tar xzvf riscv64-elf-ubuntu-22.04-gcc-nightly-2024.08.06-nightly.tar.gz
mv riscv riscv-22.04

ldd -v riscv-22.04/bin/riscv64-unknown-elf-gcc
        linux-vdso.so.1 (0x00007ffc443f6000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fee7ab5b000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fee7ad8f000)

        Version information:
        riscv-22.04/bin/riscv64-unknown-elf-gcc:
                ld-linux-x86-64.so.2 (GLIBC_2.3) => /lib64/ld-linux-x86-64.so.2
                libc.so.6 (GLIBC_2.14) => /lib/x86_64-linux-gnu/libc.so.6
                libc.so.6 (GLIBC_2.32) => /lib/x86_64-linux-gnu/libc.so.6
                libc.so.6 (GLIBC_2.3) => /lib/x86_64-linux-gnu/libc.so.6
                libc.so.6 (GLIBC_2.33) => /lib/x86_64-linux-gnu/libc.so.6
                libc.so.6 (GLIBC_2.7) => /lib/x86_64-linux-gnu/libc.so.6
                libc.so.6 (GLIBC_2.11) => /lib/x86_64-linux-gnu/libc.so.6
                libc.so.6 (GLIBC_2.34) => /lib/x86_64-linux-gnu/libc.so.6
                libc.so.6 (GLIBC_2.3.4) => /lib/x86_64-linux-gnu/libc.so.6
                libc.so.6 (GLIBC_2.9) => /lib/x86_64-linux-gnu/libc.so.6
                libc.so.6 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libc.so.6
                libc.so.6 (GLIBC_2.4) => /lib/x86_64-linux-gnu/libc.so.6
        /lib/x86_64-linux-gnu/libc.so.6:
                ld-linux-x86-64.so.2 (GLIBC_2.2.5) => /lib64/ld-linux-x86-64.so.2
                ld-linux-x86-64.so.2 (GLIBC_2.3) => /lib64/ld-linux-x86-64.so.2
                ld-linux-x86-64.so.2 (GLIBC_PRIVATE) => /lib64/ld-linux-x86-64.so.2
```
So, the 20.04 build needs GLIBC >= 2.14 while the 22.04 build needs GLIBC >= 2.33.

If GLIBC or other dependencies are a problem for your then maybe the xPack binary tools might be a better option as they are built to be as self contained as possible and just require a minimum GLIBC version (currently 2.27 or later):

- https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/
  - https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/blog/2024/08/07/riscv-none-elf-gcc-v14-2-0-1-released/#prerequisites

Note that the xPack toolchain uses a different prefix string to `riscv-gnu-toolchain`:

- https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/docs/user/#riscv-none-elf-gcc

amithmath, 2024-08-18:
I tried to compile with this:  riscv32-elf-ubuntu-20.04-gcc-nightly-2024.08.06-nightly.tar.gz  like: riscv/bin/riscv32-unknown-elf-gcc -march=rv32gc_zfh test.c -c

I am getting following error:
riscv/bin/../libexec/gcc/riscv32-unknown-elf/13.2.0/cc1: error while loading shared libraries: libmpfr.so.6: cannot open shared object file: No such file or directory

When  I tried to install libmpfr.so.6 in CentOS 7, I am getting following error:

error: Failed dependencies:
	rpmlib(PayloadIsZstd) <= 5.4.18-1 is needed by mpfr-4.1.0-7.el9.x86_64

Bur there is no rpmlib(PayloadIsZstd) 5.4.18-1 for CentOS 7

TommyMurphyTM1234, 2024-08-18:
> I tried to compile with this: riscv32-elf-ubuntu-20.04-gcc-nightly-2024.08.06-nightly.tar.gz like: riscv/bin/riscv32-unknown-elf-gcc -march=rv32gc_zfh test.c -c
>
> I am getting following error: [...]

These are general dependency issues and not `riscv-gnu-toolchain` specific issues.
You'll need to figure them out yourself or use a more up to date Linux distro and/or compile the toolchain from scratch according to your needs or use the xPack pre-built tools.

Actually, just checking the latest xPack release multilibs and I'm not sure that they include `Zfh` support:

- https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/blog/2024/08/07/riscv-none-elf-gcc-v14-2-0-1-released/#supported-libraries

I would suggest that you do the following on an up to date Linux distro:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-993
cd issue-993
./configure --prefix=`pwd`/installed-tools --with-arch=rv32gc_zfh --with-abi=ilp32d
make 2>&1 | tee build.log
```
assuming that the selected `arch/abi` are what you need to test with - adjust them if necessary.

TommyMurphyTM1234, 2024-08-18:
> I guess you meant pre-built CI toolchain binaries from: wget https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2024.08.06/riscv64-elf-ubuntu-22.04-gcc-nightly-2024.08.06-nightly.tar.gz tar xzvf https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2024.08.06/riscv64-elf-ubuntu-22.04-gcc-nightly-2024.08.06-nightly.tar.gz cd riscv/bin ? I did try riscv/bin/riscv64-unknown-gcc -march=rv64gc_zfh test.c -c I am getting following error: /lib64/libc.so.6: version `GLIBC_2.32' not found (required by riscv/bin/riscv64-unknown-elf-gcc)
> 
> If not, please point me pre-built CI toolchain binaries
> I guess you meant pre-built CI toolchain binaries from: wget https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2024.08.06/riscv64-elf-ubuntu-22.04-gcc-nightly-2024.08.06-nightly.tar.gz tar xzvf https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2024.08.06/riscv64-elf-ubuntu-22.04-gcc-nightly-2024.08.06-nightly.tar.gz cd riscv/bin ? I did try riscv/bin/riscv64-unknown-gcc -march=rv64gc_zfh test.c -c I am getting following error: /lib64/libc.so.6: version `GLIBC_2.32' not found (required by riscv/bin/riscv64-unknown-elf-gcc)
> 
> If not, please point me pre-built CI toolchain binaries

I did a build as follows (on Ubuntu 22.04.4 on WSL2) and it worked fine:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-993
cd issue-993
./configure --prefix=`pwd`/installed-tools --with-arch=rv32gc --with-abi=ilp32 --disable-gdb
make 2>&1 | tee build.log

...

./installed-tools/bin/riscv32-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=./installed-tools/bin/riscv32-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/user/Downloads/issue-993/installed-tools/libexec/gcc/riscv32-unknown-elf/13.2.0/lto-wrapper
Target: riscv32-unknown-elf
Configured with: /home/user/Downloads/issue-993/gcc/configure --target=riscv32-unknown-elf --prefix=/home/user/Downloads/issue-993/installed-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion= --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/user/Downloads/issue-993/installed-tools/riscv32-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=ilp32 --with-arch=rv32gc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 13.2.0 ()
```

amithmath, 2024-08-18:
It seems  I have issue on linux-distros...but does riscv32-unknown-elf-gcc supports _Float8?

TommyMurphyTM1234, 2024-08-18:
> It seems I have issue on linux-distros...but does riscv32-unknown-elf-gcc supports _Float8?

Did you read my earlier post? 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/993#issuecomment-2295234064

`_Float8` doesn't seem to be a GCC float type.

- https://gcc.gnu.org/onlinedocs/gcc/Floating-Types.html

amithmath, 2024-08-18:
Thanks got it.

TommyMurphyTM1234, 2024-08-18:
Closing this since the original issue is moot now that `Zfh` extension has been ratified and support implemented in the toolchain and subsequent questions have been addressed.

closed 2024-08-18 by TommyMurphyTM1234