riscv-collab/riscv-gnu-toolchain #995: Matrix Multiplication Library

Kian75, 2021-12-01:
Hi I have been working on Convolutional Neural Networks recently on RiscV cores... I have noticed that there are some libraries under the name of BLAS(basic linear algebra) that are for matrix multiplication and vector operations which can help a lot to accelerate deep learning applications in hardware... Is there any similar library in your toolchain covering this feature?

jim-wilson, 2021-12-01:
This is outside the scope of riscv-gnu-toolchain, which is only packaging for the GNU toolchain.  There may be ports of various libraries that might be useful to you, but we don't cover them here.  Maybe try asking on the public riscv.org sw-dev mailing list.  

Also see the RISC-V exchange on the riscv.org site which has lists of available libraries.  There is a pointer to a BLAS port, and TensorFlow Lite, and some other stuff.  Not everything is listed here though, only stuff that has been registered with the RISC-V Exchange would be listed.
[https://riscv.org/exchange/software/](https://riscv.org/exchange/software/l)
Try the ML&AI and Accelerated Libraries tabs.

Kian75, 2021-12-02:
Thank you for your answer @jim-wilson ...

TommyMurphyTM1234, 2022-09-27:
Not a RISC-V GCC toolchain issue.

closed 2022-09-27 by TommyMurphyTM1234