riscv-collab/riscv-gnu-toolchain #996: How can I link this file in this situation?

naveen126, 2021-12-07:
HI Sir,
Here is a simple example for this question.

In riscv.ld file:
    . = 0x10000;
	.tmp_text : 
	{
		*(.tmp_text)
		*(.tmp_text.*)
	}

       . = 0x20000;
	.text : 
	{
		*(.text)
		*(.text.*)
	}

In example.S file, there are some code in .section.tmp_text and also some code in .section.text.

First I can build example.S into example.o.

I want a elf file without .section.tmp_text.
But I don't know how to link it? I do not know the options for riscv64-unknown-elf-ld to do that.

Please give me some help! Thanks a lot!

jim-wilson, 2021-12-07:
You can use /DISCARD/ is a linker script to discard sections at link time.  However, I would suggest that the easiest way is to remove the code you don't want from example.S. You could use conditional compilation via #if for instance.

Linker documentation
[https://sourceware.org/binutils/docs-2.37/ld/Output-Section-Discarding.html#Output-Section-Discarding](https://sourceware.org/binutils/docs-2.37/ld/Output-Section-Discarding.html#Output-Section-Discarding)

Linux kernel example
[https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/arch/mips/vdso/vdso.lds.S?h=v5.16-rc4#n60](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/arch/mips/vdso/vdso.lds.S?h=v5.16-rc4#n60)

TommyMurphyTM1234, 2022-09-27:
GNU LD question, not an issue with the RISC-V GCC toolchain.
Closing.

closed 2022-09-27 by TommyMurphyTM1234