riscv-collab/riscv-gnu-toolchain #997: Need help on compiler option that helps map ISA spec version with RISC-V extension

Sharanabasavaexaleap, 2021-12-10:
Hello,
I need to map extensions to specific ISA in final application exe creation.
For example:
◦ RV32I - version 2.1
◦ RVM - version 2.0
- RVC - version 2.2

Tried option "-march=rv32imacfd-2.0" but it does not work.

A) ELF toolchain build
I have built riscv ELF toolchain using following commands
./configure --prefix=/home/riscv-gnu-toolchain/mytool --with-arch=rv32gc --with-abi=ilp32d
make 
 
B)
Compile application test.c
home/mytool/bin/riscv32-unknown-elf-gcc -march=rv32imacfd-2.0 hello.c -o hello.out
Error:
riscv32-unknown-elf-gcc: error: '-march=rv32imacfd-2.0': ISA string is not in canonical order. 'f'

Can you please help fix?

Thanks
[image]

TommyMurphyTM1234, 2021-12-10:
Did you read section _27.4 Version Numbers_ of the RISC-V ISA Unprivileged specification for details of how extension version numbers are formatted? 

https://riscv.org/technical/specifications/

I'm not sure offhand how exhaustive the support is in the tools right now for parsing such strings but it should be easy to just try it.

jim-wilson, 2021-12-10:
The compiler expands the user -march command into long form for the arch attribute.  So you can just check the .s file to see what is going on.

```
rohan:2022$ riscv32-unknown-elf-gcc -O -S tmp.c
rohan:2023$ grep arch tmp.s
	.attribute arch, "rv32i2p0_m2p0_a2p0_f2p0_d2p0_c2p0"
rohan:2024$ riscv32-unknown-elf-gcc -O -S -march=rv32i2p1_m2p0_c2p2 tmp.c
cc1: error: requested ABI requires '-march' to subsume the 'D' extension
rohan:2025$ riscv32-unknown-elf-gcc -O -S -march=rv32i2p1_m2p0_c2p2 tmp.c -mabi=ilp32
rohan:2026$ grep arch tmp.s
	.attribute arch, "rv32i2p1_m2p0_c2p2"
rohan:2027$ 
```

Sharanabasavaexaleap, 2021-12-21:
> The compiler expands the user -march command into long form for the arch attribute. So you can just check the .s file to see what is going on.
>
> ```
> rohan:2022$ riscv32-unknown-elf-gcc -O -S tmp.c [...]

Hello Jim
Thanks for inputs.
Tried following
**Makefile.in modification:**
Compiler opions " -march=rv32i2p0_m2p0_a2p0_f2p0_d2p0_c2p0 -mabi=ilp32d"  are added to CFLAGS CXXFLAGS to build newlib and newlib nano libraries

**Toolchain build:**
./configure --prefix=<> --with-arch=rv32imafdc 
make

The toolchain got successfully built.

**Smoke test**
../new_install/bin/riscv32-unknown-elf-gcc hello.c -o hello.out -march=rv32i2p1m2p2af2p2d2p2c2p0 -mabi=ilp32d
/home/deepa/work/GF2_code_play/new_install/lib/gcc/riscv32-unknown-elf/11.1.0/../../../../riscv32-unknown-elf/bin/ld: warning: /home/deepa/work/GF2_code_play/new_install/lib/gcc/riscv32-unknown-elf/11.1.0/crtbegin.o: mis-matched ISA version 2.0 for 'i' extension, the output version is 2.1
/home/deepa/work/GF2_code_play/new_install/lib/gcc/riscv32-unknown-elf/11.1.0/../../../../riscv32-unknown-elf/bin/ld: warning: /home/deepa/work/GF2_code_play/new_install/lib/gcc/riscv32-unknown-elf/11.1.0/crtbegin.o: mis-matched ISA version 2.0 for 'm' extension, the output version is 2.2
/home/deepa/work/GF2_code_play/new_install/lib/gcc/riscv32-unknown-elf/11.1.0/../../../../riscv32-unknown-elf/bin/ld: warning: /home/deepa/work/GF2_code_play/new_install/lib/gcc/riscv32-unknown-elf/11.1.0/crtbegin.o: mis-matched ISA version 2.0 for 'f' extension, the output version is 2.2
/home/deepa/work/GF2_code_play/new_install/lib/gcc/riscv32-unknown-elf/11.1.0/../../../../riscv32-unknown-elf/bin/ld: warning: /home/deepa/work/GF2_code_play/new_install/lib/gcc/riscv32-unknown-elf/11.1.0/crtbegin.o: mis-matched ISA version 2.0 for 'd' extension, the output version is 2.2
/home/deepa/work/GF2_code_play/new_install/lib/gcc/riscv32-unknown-elf/11.1.0/../../../../riscv32-unknown-elf/bin/ld: warning: /home/deepa/work/GF2_code_play/new_install/lib/gcc/riscv32-unknown-elf/11.1.0/crtend.o: mis-matched ISA version 2.0 for 'i' extension, the output version is 2.1
/home/deepa/work/GF2_code_play/new_install/lib/gcc/riscv32-unknown-elf/11.1.0/../../../../riscv32-unknown-elf/bin/ld: warning: /home/deepa/work/GF2_code_play/new_install/lib/gcc/riscv32-unknown-elf/11.1.0/crtend.o: mis-matched ISA version 2.0 for 'm' extension, the output version is 2.2
/home/deepa/work/GF2_code_play/new_install/lib/gcc/riscv32-unknown-elf/11.1.0/../../../../riscv32-unknown-elf/bin/ld: warning: /home/deepa/work/GF2_code_play/new_install/lib/gcc/riscv32-unknown-elf/11.1.0/crtend.o: mis-matched ISA version 2.0 for 'f' extension, the output version is 2.2
/home/deepa/work/GF2_code_play/new_install/lib/gcc/riscv32-unknown-elf/11.1.0/../../../../riscv32-unknown-elf/bin/ld: warning: /home/deepa/work/GF2_code_play/new_install/lib/gcc/riscv32-unknown-elf/11.1.0/crtend.o: mis-matched ISA version 2.0 for 'd' extension, the output version is 2.2

There are spec version mismatch warnings with crtbegin.o and crtend.o

**Questions**
1) Are there any quick ways to fix Makefile.in to resolve crtbegin.o and crtend.o warnings?
2) Do you know if spike / pk can simulate instructions for different extensions having different ISA version ?

Thanks,

We are able to build newlib with the respective ISA version and extension :rv32i2p0_m2p0_a2p0_f2p0_d2p0_c2p0 
Updated CFLAG
[Detailed log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/7755884/Detailed.log)
S

TommyMurphyTM1234, 2021-12-21:
> Tried following **Makefile.in modification:** Compiler opions " -march=rv32i2p0_m2p0_a2p0_f2p0_d2p0_c2p0 -mabi=ilp32d" are added to CFLAGS CXXFLAGS to build newlib and newlib nano libraries
> 
> **Toolchain build:** ./configure --prefix=<> --with-arch=rv32imafdc make

Does configure take a fully versioned architecture string?
E.g.

`./configure ... --with-arch=rv32i2p0_m2p0_a2p0_f2p0_d2p0_c2p0`

jim-wilson, 2021-12-21:
I don't follow spike, so I don't know if it supports different ISA versions.

To eliminate the startfile warnings, you either need to configure with the right ISA, as Tommy Murphy mentioned, or multilib your toolchain, i.e. build newlib twice with different ISA versions.  We currently don't multilib on ISA versions, so you would have to set up your own multilib list for that.  See the --with-multilib-generator info in the README.md file.

Unfortunately, using a configure --with-arch= option with ISA versions doesn't work as is because the arch-canonicalize script doesn't handle ISA versions.  However, it is trivial to disable that, either in the script or in riscv-gcc/gcc/config.gcc where it is called, and then I get a successful build.  The script should be fixed.  Kito can probably best do that.

Trying --with-multilib-generator, I see that the multilib-generator script also calls arch-canonicalize so it also fails.  But if I hack around that to disable the calls, then this works also.

Sharanabasavaexaleap, 2021-12-22:
> > Tried following **Makefile.in modification:** Compiler opions " -march=rv32i2p0_m2p0_a2p0_f2p0_d2p0_c2p0 -mabi=ilp32d" are added to CFLAGS CXXFLAGS to build newlib and newlib nano libraries
> > **Toolchain build:** ./configure --prefix=<> --with-arch=rv32imafdc make
> 
> I wonder if configure can accept a fully versioned architecture string? E.g.
> 
> `./configure ... --with-arch=rv32i2p0_m2p0_a2p0_f2p0_d2p0_c2p0`

No, it will not accept, if we pass `./configure ... --with-arch=rv32i2p0_m2p0_a2p0_f2p0_d2p0_c2p0 this we get error like configure-gcc failed.

TommyMurphyTM1234, 2021-12-22:
> No, it will not accept, if we pass `./configure ... --with-arch=rv32i2p0_m2p0_a2p0_f2p0_d2p0_c2p0 this we get error like configure-gcc failed.

Ok, so you probably need to do what @jim-wilson suggests above and hack the configure script to not call arch-canonicalize.

Sharanabasavaexaleap, 2022-01-14:
Hi jim wilson, 
I want to build tool chain for 10.2.0, what procedure/steps i have to follow to get 10.2.0 version. From gnu tool chain only i can get 11.0.

Thanks in advance,
Sharan

TommyMurphyTM1234, 2022-01-14:
> Hi jim wilson, I want to build tool chain for 10.2.0, what procedure/steps i have to follow to get 10.2.0 version. From gnu tool chain only i can get 11.0.
> 
> Thanks in advance, Sharan

You can get a 10.2.0 build from here if that helps and obviates the need for your to build it yourself?

https://xpack.github.io/riscv-none-embed-gcc/#releases
https://xpack.github.io/riscv-none-embed-gcc/releases/

Sharanabasavaexaleap, 2022-01-14:
> > Hi jim wilson, I want to build tool chain for 10.2.0, what procedure/steps i have to follow to get 10.2.0 version. From gnu tool chain only i can get 11.0.
> > Thanks in advance, Sharan
> 
> You can get a 10.2.0 build from here if that helps and obviates the need for your to build it yourself?
>Thanks your reply,
i think you already know that from issues i rised. i am trying to build toolchain using different isa spec version and i am getting toolchain gcc 11.1.0. now i want it for 10.2.0. so is any think i have pass while cloning or in make command or above links you shared is that enough?

TommyMurphyTM1234, 2022-01-14:
> i just need for newlib(elf) 32bit only.

Whether the build of the RISC-V bare metal/newlib toolchain has 32 or 64 (bit) in its name it can build both 32 and 64 bit software.

ilg-ul, 2022-01-14:
> Whether the build of the RISC-V bare metal/newlib toolchain has 32 or 64 (bit) in its name it can build both 32 and 64 bit software.

Exactly to avoid this common confusion, the **xPack RISC-V Embedded GCC** uses the more generic `riscv-none-embed-` prefix.

jim-wilson, 2022-01-17:
We only support one gcc version at a time, generally the last one, a few months after it was released.  However, since the sources are in git, you can check out older versions to get older compilers.  They just aren't supported.  So do a git log on the source tree piping output to a file, and then look for the patch that updated gcc to gcc 11.  That happened in June 2021.  There were binutils and qemu updates in April 2021.  A glibc update in February.  Binutils updates are usually safe, so taking the last commit in April before we started the gcc related changes in June should be OK.
```
commit b715e4f01b43efef487166f75d5d85d3c33fa7ef
Author: Kito Cheng <kito.cheng@sifive.com>
Date:   Thu Apr 22 23:32:35 2021 +0800

    Bump riscv-binutils

```
You can just do "git checkout b715e4f01b43efef487166f75d5d85d3c33fa7ef".  And then don't forget the "git submodule update --init --recursive".  This works best if you check out the right commitid before you update submodules, otherwise you may have to manually clean up the submodules to deal with conflicts.

TommyMurphyTM1234, 2023-10-15:
> No, it will not accept, if we pass `./configure ... --with-arch=rv32i2p0_m2p0_a2p0_f2p0_d2p0_c2p0 this we get error like configure-gcc failed.

This no longer fails with the latest `riscv-gnu-toolchain` commit:

* https://github.com/riscv-collab/riscv-gnu-toolchain/commit/6b1324367b879a9b89437846827b48151b26b412

```
./configure --prefix=`pwd`/installed-tools --with-arch=rv32i2p0_m2p0_a2p0_f2p0_d2p0_c2p0 --with-abi=ilp32d
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for fgrep... /usr/bin/grep -F
checking for grep that handles long lines and -e... (cached) /usr/bin/grep
checking for bash... /bin/bash
checking for __gmpz_init in -lgmp... yes
checking for mpfr_init in -lmpfr... yes
checking for mpc_init2 in -lmpc... yes
checking for curl... /usr/bin/curl
checking for wget... /usr/bin/wget
checking for ftp... /usr/bin/ftp
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed
```
As such, I am closing this issue off as resolved.
If anybody thinks that there is still an issue please post and/or reopen the issue.

closed 2023-10-15 by TommyMurphyTM1234