riscv-collab/riscv-gnu-toolchain #998: Error while using boost library

Karthiknimit, 2021-12-16:
Hello!

I have been using GCC to to generate elf for RISC-V  rv32i and it all works fine with the below compile instruction. `/opt/riscv/bin/riscv64-unknown-elf-gcc -c -march=rv32i  -mabi=ilp32 -g3 -O0  -MD -fstrict-volatile-bitfields -fno-exceptions  -I../include/ -o build/obj/src/main.o src/main.cpp`

Now in have to include boost libraries in my main.cpp. So I included below line in main.cpp 
`#include <boost/units/quantity.hpp>`

I got to know GCC doesn't add C++ libraries. To do so -lstdc++ needs to added during invocation and so I tried with below instruction.
`/opt/riscv//bin/riscv64-unknown-elf-gcc -c -march=rv32i  -mabi=ilp32 -g3 -O0  -MD -fstrict-volatile-bitfields -fno-exceptions 
**-lstdc++** -I../include/ -o build/obj/src/main.o src/main.cpp `
But this gives me the error 
src/main.cpp:26:11: fatal error: boost/units/quantity.hpp: No such file or directory

Without cross compiling I just tried `gcc main.cpp -lstdc++` and it worked fine. What am I missing while cross compiling?

TommyMurphyTM1234, 2021-12-16:
Boost is not part of the C++ standard library and is not built as part of the gcc/g++ toolchain. If you want to use it then you need to build it yourself. The error is clearly saying that the Boost header is not available.

https://www.boost.org/

Karthiknimit, 2021-12-16:
@TommyMurphyTM1234 [boost](https://www.boost.org/doc/libs/1_78_0/more/getting_started/unix-variants.html) says the boost libraries does not need a separate build as they are header only files. Below are the quoted words.

> Most Boost libraries are header-only: they consist entirely of header files containing templates and inline functions, and require no separately-compiled library binaries or special treatment when linking.

_gcc main.cpp -lstdc++ works on unix without any problem. When I try to cross-compile for risc-V I get the error which I have mentioned. I tried by providing the path of boost libraries with -I option. But this leads to to another error as given below.
/opt/riscv/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: build/libbla-demo.elf: hidden symbol `__dso_handle' isn't defined
/opt/riscv/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status_

cmuellner, 2021-12-16:
If you compile C++ code, then use `g++` instead of `gcc`.

For me, it is not clear, where you have the headers from. Please ensure, that you are not setting your include directory to your host system's include directory (`/usr/include`) when cross-compiling.

TommyMurphyTM1234, 2021-12-16:
> @TommyMurphyTM1234 [boost.org](https://www.boost.org/doc/libs/1_78_0/more/getting_started/unix-variants.html) says the boost libraries does not need a separate build as they are header only files. Below are the quoted words.
> 
> > The first thing many people want to know is, “how do I build Boost?” The good news is that often, there's nothing to build.
> > Nothing to Build?
> > Most Boost libraries are header-only: they consist entirely of header files containing templates and inline functions, and require no separately-compiled library binaries or special treatment when linking.

Most. Not "all".

jim-wilson, 2021-12-16:
Linux distros have teams of people to build stuff.  With riscv-gnu-toolchain, you only get what you built yourself, and if it isn't part of binutils/gcc/gdb/newlib then it isn't available unless you build it yourself.  Even if boost is only header files, you still need to fetch the package, build it, and install it in the right place so the compiler can find it.

Not everything that works on linux will work with an embedded elf compiler.  The embedded elf toolchain doesn't support shared libraries for instance.  Also, newlib doesn't support all POSIX functions that a glibc linux system supports.

dso_handle is for shared library support.  A newlib toolchain doesn't do anything with it, but there is a dummy definition in crtbegin.o to satisfy uses, and crtbegin.o is linked by default.  However, gcc-8.3 is pretty old, we are at gcc-11 now.  Maybe this is a problem with your old toolchain, or maybe it is a problem with what you are trying to do.  It isn't clear.

Karthiknimit, 2021-12-17:
@jim-wilson Thank you for the explanation. I am fairly new to the compiler world. If you can briefly explain or share some links on how to build this library, it will be very helpful to me.

TommyMurphyTM1234, 2021-12-17:
> @jim-wilson Thank you for the explanation. I am fairly new to the compiler world. If you can briefly explain or share some links on how to build this library, it will be very helpful to me.

The Boost website that I linked to earlier has full details on how to obtain, build and use these libraries. You need to read those instructions.

jim-wilson, 2021-12-17:
I don't know anything about building or using boost.

closed 2022-01-10 by Karthiknimit