riscv-collab/riscv-gnu-toolchain #999: failed to rescue into submodule path 'qemu'

Kian75, 2021-12-19:
Hi I have been trying to install vector support for your compiler with `git clone https://github.com/riscv/riscv-gnu-toolchain -b rvv-intrinsic` but after I `cd riscv-gnu-toolchain`  then `git submodule update --init --recursive` I encountered following error:
fatal: unable to access 'https://git.qemu.org/git/capstone.git/': server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.ctrl CRLfile: none 
fatal: clone of 'https://git.qemu.org/git/capstone.git' into submodule path 'capstone' failed
Failed to recurse into submodule path 'qemu'

I know you have covered similar error in [https://github.com/riscv-collab/riscv-gnu-toolchain/issues/480](url) but I still couldn't solve it... now it gives certificate error can you help?

jim-wilson, 2021-12-19:
It is working for me.  It may be a temporary problem that was already fixed. Or it may be a problem with ssl certs on your system.  I can find some answers with a web search on how to look at and fix this.
[https://fabianlee.org/2019/01/28/git-client-error-server-certificate-verification-failed/](https://fabianlee.org/2019/01/28/git-client-error-server-certificate-verification-failed/)
[https://stackoverflow.com/questions/35821245/github-server-certificate-verification-failed](https://stackoverflow.com/questions/35821245/github-server-certificate-verification-failed)

You can build the toolchain without qemu.  If you manually submodule update every dir except qemu then you will have enough to build the toolchain.  You just won't be able to run "make check" unless you specify an alternate simulator.

Kian75, 2021-12-19:
Thanks for your answer @jim-wilson as always very fast an accurate answers... by the way in  `git clone https://github.com/riscv/riscv-gnu-toolchain -b rvv-intrinsic` which vector isa is supported? I need 0.1 which is the oldest one I think but is it updated to version 0.9 which is currently the most up to date version of vector spec?

jim-wilson, 2021-12-19:
Sorry, I forgot to answer the vector question.  The rvv-intrinsics branch is not actively supported, and is not in general recommended.  All rvv work moved to LLVM a while ago, and only llvm has properly maintained rvv support.  I don't know what version of the vector spec the rvv-intrinsic branch supports, it is whatever version of the vector spec was current when the work was abandoned.

Alibaba claims to have a rvv 1.0 gcc port, but apparently has only made binaries available.  The link doesn't seem to work for me here in the US.  See [https://github.com/riscv-collab/riscv-gcc/issues/320](https://github.com/riscv-collab/riscv-gcc/issues/320)

jim-wilson, 2021-12-20:
As far as I know, neither gcc nor llvm has any autovectorization support.  If you want vector code, you have to use the intrinsics.
[https://github.com/riscv-non-isa/rvv-intrinsic-doc](https://github.com/riscv-non-isa/rvv-intrinsic-doc)

GCC has a feature where you can use regular arithmetic on vector types.  See for instance
[https://github.com/riscv-collab/riscv-gcc/blob/riscv-gcc-10.1-rvv-dev/gcc/testsuite/gcc.target/riscv/rvv/rvv_example5.c](https://github.com/riscv-collab/riscv-gcc/blob/riscv-gcc-10.1-rvv-dev/gcc/testsuite/gcc.target/riscv/rvv/rvv_example5.c)
which only uses intrinsics for vsetvl and divide.  I think LLVM does not support this feature though, and the gcc rvv port is not at rvv 1.0, so not very useful.

Kian75, 2021-12-26:
Sorry for bothering you again.. but I tried to install `git clone https://github.com/riscv/riscv-gnu-toolchain -b rvv-intrinsic` after I manually submodule update every directory except qemu, I tried `sudo make` but it gave me this error
[image]
how can I solve it?

jim-wilson, 2021-12-26:
There isn't enough info here for me to tell what is wrong.  But usually the "killed" message indicates a system resource issue, e.g. maybe you ran out of virtual memory during the build and the kernel had to kill a random process to free up some memory.  The free command will tell you how much free memory you have.  If you run the build in one window and run free repeatedly in the other window you might see memory free drop to 0 and then the cc1plus process killed.  Adding more swap space can solve this.  Or killing some other process using a lot of memory.

It is a known problem that building the rvv compiler port needs a lot of memory.  The V extension adds about 7000 instruction patterns to the riscv.md file, which is a lot more than any other extension, and probably more than every other extension combined.  It may not be possible to build it on a 32-bit host system limited to 4GB of virtual memory.

Kian75, 2021-12-27:
Thank you @jim-wilson even though I have 8GB RAM and 7.5GB for my VMWARE it still gives the same error...
Anyway... thank you for your answer

jim-wilson, 2021-12-27:
I think you need more memory than that.  You can create a swap file to get more virtual memory.

TommyMurphyTM1234, 2022-09-27:
These seem to be questions about build issues and RISC-V tools vector support rather than issues with the RISC-V GCC toolchain itself.