riscv-collab/riscv-gnu-toolchain #1000: Need that helps to map new  ISA spec version with RISC-V extension  for rv32

Sharanabasavaexaleap, 2021-12-20:
Hi , For theese ISA spec versions for newlib  
  RV32I - version 2.1
◦ RVM - version 2.0
◦ RVC - version 2.0
◦ RVF - version 2.2
◦ RVD - version 2.2
i am getting an error :
--with-abi=ilp32d2p2 is not supported
CXX posix-hdep.o
Makefile:4368: recipe for target 'configure-gcc' failed
make[1]: *** [configure-gcc] Error 1
make[1]: Leaving directory '/home/riscv-gnu-toolchain/build-gcc-newlib-stage1'
Makefile:517: recipe for target 'stamps/build-gcc-newlib-stage1' failed
make: *** [stamps/build-gcc-newlib-stage1] Error 2 

1.for configure i passed : 
./configure --prefix=/home/riscv-gnu-toolchain/bin/  --with-arch=rv32i2p1_m_f2p2_d2p2_c  --with-abi=ilp32d2p2
* New spec extensions version given as by referring riscv isa spec manual 
2. riscv32-unkonwn-elf-gcc is missing when i pass these versions with arch.

kito-cheng, 2021-12-20:
ABI should still use `--with-abi=ilp32d` even `d` using `2.2`, ABI are contemptible between different ISA versions :)

cmuellner, 2021-12-20:
What is a `ilp32d2p2` ABI?
The [README.md](https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/README.md) lists the supported ABIs:
Supported ABIs are ilp32 (32-bit soft-float), ilp32d (32-bit hard-float), ilp32f (32-bit with single-precision in registers and double in memory, niche use only), lp64 lp64f lp64d (same but with 64-bit long and pointers).

At the moment there is no need to pass the ISA specification version as part of the arch string.

Independent of that:
The default value for the arch string is `rv64imafdc`. Given your list of ISA specs, I assume this is what you want:
`./configure --prefix=/opt/rv32-elf --with-arch=rv32gc --with-abi=ilp32d && make -j$(nproc)`
After this command terminates, you will find your gcc here: `/opt/rv32-elf/bin/riscv32-unknown-elf-gcc`.

Sharanabasavaexaleap, 2021-12-21:
> What is a `ilp32d2p2` ABI? The [README.md](https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/README.md) lists the supported ABIs: Supported ABIs are ilp32 (32-bit soft-float), ilp32d [...]

i need to pass new isa spec versions, for that what is correct way to get without error?

jim-wilson, 2021-12-21:
The -mabi option does not accept versions.  The -march option does.  ISA versions are specified in the -march option.

Tools may not accept new ISA versions if support for them hasn't been added yet.  You need to allow at least a few months after a new ISA version is ratified before it is available in development sources, and a few more months before it appears in a new release.  GCC only makes releases once a year in the spring, and the new stuff was just approved, so it might be a year and a half before it appears in an official GCC release.

TommyMurphyTM1234, 2022-09-29:
No issue with the RISC-V GCC toolchain itself identified here.

closed 2022-09-29 by TommyMurphyTM1234