riscv-collab/riscv-gnu-toolchain #1001: How to use LLVM+Clang 13.0 vector extension

CodeDance364, 2021-12-21:
I have installed the LLVM+Clang 13.0,which is abtained from release.llvm.org,but it comes out errors showing can't recognize the instruction when I compile vector instructions such as VSETVLI,but none-vector code can be compiled correctlly.So I want to comfirm several questions:
1.Can LLVM+Clang 13.0 support vector extension compiling?
2.If yes,how to use the vector feature?It is controled by configuration while compiling the LLVM+Clang,or by parameter while run Clang to compile the application code?
3.I found that using -menable-experimental-extensions and -march=rv64gcv0p10 is ok,but -march=rv64gcv1p0 comes out error showing not support 1.0 but only support 0.10.But in my understanding,the latest vector extension is 1.0.Which vector version is the rv64gcv0p10 corrsponding to?

Thanks!

jim-wilson, 2021-12-21:
I don't follow LLVM.  But I note that rvv 1.0 was only just recently approved, and not yet merged into the ISA document.  It isn't reasonable to expect that an existing llvm release would support a standard that didn't exist until after the release.  It takes time to write and review software, so you need to allow at least a few months for stuff to appear in llvm development sources, and even longer for it to appear in a release.

v0p10 is effectively what was proposed for the rvv 1.0 standard, but there might have been a few minor changes before it was approved, probably none of which effect the compiler.  Someone will have to check though.

Searching phabricator for rvv, I found this
[https://reviews.llvm.org/D112987](https://reviews.llvm.org/D112987)

There aren't many llvm people watching this gnu toolchain git repo.  You might try a public riscv.org mailing list like sw-dev.  Or maybe some llvm mailing list.  Or attend a meeting.  There are bi-weekly meetings for LLVM RISC-V developers.
[https://github.com/llvm/llvm-project/blob/main/llvm/docs/GettingInvolved.rst#id42](https://github.com/llvm/llvm-project/blob/main/llvm/docs/GettingInvolved.rst#id42)

CodeDance364, 2021-12-21:
Thanks for your quick reply,I think you've helped me solve the question3,which means v0p10 is exactly the effective one that we can use now.
For further LLVM,maybe I should go to LLVM project repo.
Thank you very much.

closed 2022-09-27 by TommyMurphyTM1234