riscv-collab/riscv-gnu-toolchain #1003: Adding rvv intrinsic to already installed toolchain

Mousavikia, 2021-12-26:
Hi I want to install rvv intrinsic to already installed compiler is it possible?
What I mean is I already installed gnu toolchain with this configs:
`git clone https://github.com/riscv/riscv-gnu-toolchain`
`./configure --prefix=/opt/riscv --enable-multilib`
`make`
Can I reconfigure or do something else to have rvv intrinsics on top of it rather that installing rvv intrinsic from scratch like this:
`git clone https://github.com/riscv/riscv-gnu-toolchain -b rvv-intrinsic`
`cd riscv-gnu-toolchain`
`git submodule update --init --recursive`
`mkdir build && cd build`
`../configure --prefix=/desired/installation/path`
`make`

TommyMurphyTM1234, 2021-12-26:
As far as I know if you want RVV then you should be looking at llvm as RVV work on GCC (although maybe not binutils) effectively stopped a good while ago. See here for example.

https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1001
https://github.com/riscv-collab/riscv-gnu-toolchain/issues/975

jim-wilson, 2021-12-26:
No, you can't add a feature to an already installed toolchain.  You have to build a new toolchain from scratch.  And as Tommy Murphy mentioned, the gcc rvv support is broken, incomplete, and poorly maintained.  Use LLVM instead if you want working rvv support.

Mousavikia, 2021-12-27:
Thank you @jim-wilson and @TommyMurphyTM1234 
By the way... I looked at this [https://github.com/sifive/riscv-llvm](url) for installing llvm is it ok? And do you know where can I force installation to have V instructions as well?

jim-wilson, 2021-12-27:
I would suggest starting at llvm.org.  The sifive/riscv-llvm tree is probably an out-of-date development tree.  LLVM contains V support on the mainline development tree.  I think the last release has v0.10 support, and mainline does not yet have v1.0 support last I looked, but there is a patch in phabricator for this.

fwsGonzo, 2022-02-14:
Does objdump and GDB read the vector instructions properly? I was thinking about implementing this, even if I can't test it with GCC yet.

TommyMurphyTM1234, 2022-09-27:
Not an issue with the RISC-V GCC toolchain.

closed 2022-09-27 by TommyMurphyTM1234

TommyMurphyTM1234, 2022-11-10:
This is a list of all issues logged that I can find that relate to adding new instructions to the toolchain:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1095
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1055
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/837
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/782
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1003
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/795
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/660
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/543
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/341
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/115

At this stage in the evolution of RISC-V,the majority of instruction addition will be done under the auspices of the relevant groups working on standardising, prototyping, and implementing/enhancing RISC-V extensions and their corresponding instructions. As such, there should be less of a demand for others to do so.

However, where somebody still wants to implement a non stadard custom extension/instruction here are some useful resources:

* GCC internals documentation: https://gcc.gnu.org/onlinedocs/gccint/
* https://hsandid.github.io/posts/risc-v-custom-instruction/
* https://ossg.bcs.org/blog/2020/10/29/adding-an-instruction-to-the-gnu-assembler/
* https://embdev.net/topic/228773
* https://groups.google.com/u/0/a/groups.riscv.org/g/sw-dev/c/sL_OHXYj3LY/m/NOm5Fwa3BAAJ
* https://forums.sifive.com/t/adding-custom-instruction-to-riscv-toolchain/4044/2