riscv-collab/riscv-gnu-toolchain #1004: [riscv-gdb] can I modify `mtime` register with GDB?

PAN-Ziyue, 2021-12-27:
I have tried multiple ways to change the value in `mtime` register including:

- `set * 0x200bff8 = <some value>`
- `set $time = <some value>`

But none of them succeeded. Is there a way to modify `mtime` with GDB? Thanks.

jim-wilson, 2021-12-27:
Is there actually a time register?  The ISA doesn't require this, and some hardware (like SiFive parts) don't implement it.  SiFive hardware instead takes a trap on reference and emulates it in a trap handler.  If no time register, then obviously you can't use gdb to read or write it.  GDB does support it, but this is of no use if the hardware doesn't.

Are you using a linux hosted gdb?  If a linux hosted gdb, then there is no ptrace support for csrs as yet, so no way for gdb to read/write csrs like the time register.  If using the remote protocol to connect to a target, then there is the question of whether or not the remote stub supports csrs.  It may or may not.  If using gdb with openocd, then you definitely have access to csrs, if they exist on the target.

If it is mtime specifically that you want, then gdb can't read or write it unless you are in machine mode.  Or using openocd which uses jtag.

Gdb doesn't directly support mtime because it isn't a "register".  It is a target dependent address in memory.  Gdb can't know the address as it doesn't have this kind of target info.

PAN-Ziyue, 2021-12-27:
> Is there actually a time register? The ISA doesn't require this, and some hardware (like SiFive parts) don't implement it. SiFive hardware instead takes a trap on reference and emulates it in a trap [...]

Much appreciated for your immediate reply. 

I am running a standalone elf on `qemu-system-riscv64`, with riscv-gdb as the debugger (using remote serial protocol). AFAIK, `virt` machine in qemu does support CSR and mtime (as an mmio register defined in PLIC).

jim-wilson, 2021-12-27:
qemu doesn't support writes to the time register.  You can only read it.

qemu probably doesn't have a proper time register.  It has no way to simulate one.  It probably instead uses host system time to emulate the time register, and you can't use gdb to modify the host system time this way.  So it supports reads from time but not writes to time.

If you want a time register, then you need to use source for a core and run that in a simulator like verilator.

PAN-Ziyue, 2021-12-28:
Thanks Jim!

closed 2021-12-28 by PAN-Ziyue