riscv-collab/riscv-gnu-toolchain #1005: Nightly binary fails to build anything or provide --target-help

ekliptik, 2021-12-28:
With riscv64-glibc-ubuntu-20.04-nightly-2021.09.21-nightly.tar.gz
installed on Arch Linux using aur package [riscv-gnu-toolchain](https://aur.archlinux.org/packages/riscv-gnu-toolchain/)
output of `riscv32-unknown-elf-gcc --target-help` ends with
```
Assembler options
=================

Use "-Wa,OPTION" to pass "OPTION" to the assembler.

Assembler messages:
Fatal error: invalid -march= option: `rv32imafdc'
```
Similarly, this error is reported when for example RISC-V ISA tests are built. It seems unlikely to be an Arch-specific issue

ekliptik, 2021-12-28:
I should have read the comments on the AUR repo page, it *is* Arch-specific. However, it seems odd that this GCC build tries to use the system's GCC assembler. Could this be fixed here, by calling explicitly the RISC-V assembler?

jim-wilson, 2021-12-28:
gcc calls as, with a search path that should include a dir with a riscv specific assembler, usually this is $prefix/$target/bin/as, and you don't put that dir on your path as that will break the native compiler.  You can see the gcc search path by adding a --print-search-dirs option.  If the search path is broken, or if the $prefix/$target/bin dir is missing, that is a build problem.  Adding a -v option to a gcc command will show exactly what the gcc driver is going.

Things may work slightly differently on a multiarch system like Debian/Ubuntu.  There is still a dir with target tools in it, but the path to the dir changes.  This depends on how you configure and build the compiler/cross-compiler.

Changing the name of the assembler causes problems.  What happens if you have a rv64gc toolchain and you specify -march=rv32gc.  What exactly is the name  of the assembler now?  There are also other things that break.  The easiest solution that the gcc developers found is to not change the as name, and instead always put a dir on the search path that contains the right as for the target.

I looked at the riscv32-elf-ubuntu-20.04-nightly-2021.09.16-nightly.tar.gz and it looks OK.  It has the $target/bin/as as I expect, and works for me on an ubuntu system.  Though it seems curious that "ubuntu" is in the name for what is supposed to be an Arch package.

Also curious that you mentioned the riscv64-glibc package, but are running the riscv32-elf compiler that is not in that package.  Maybe you meant the riscv32-elf package.

JakubVanek, 2021-12-29:
I think I've found the cause of the issue. The problem is indeed the Arch package (see [PKGBUILD](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=riscv-gnu-toolchain) or [repo](https://github.com/sickcodes/aur/tree/master/riscv-gnu-toolchain)). The packaging script deletes the `$prefix/$target` directory where the target-specific binutils, libc and other files reside. Nightly builds from this riscv-collab repository don't have this problem.

ekliptik, 2022-01-03:
Thank you for your thorough explanation, Jim. I am closing this issue, as it's not within this repo's scope

closed 2022-01-03 by ekliptik