riscv-collab/riscv-gnu-toolchain #1006: Where to get the version number of each extension ISA that riscv-gnu-toolchain support?

CodeDance364, 2021-12-29:
Hi,
  Where can we get the version number of each extension ISA that riscv-gnu-toolchain support,such as 0p10 in rv64gcv0p10?Since we need to make sure the ISA version that tool-chain used match with ISA version that core used,otherwise the core will not be able to decode the instructions that encoded with different ISA format.
Thanks!

jim-wilson, 2021-12-29:
The compiler expands the user specified arch string to long form, so just compile with -S -march=rv64gcv and then look at the .s file for the arch string.

GCC should not be used for V support.  Only LLVM has working V support at this time.  Upstream binutils development sources do have V support, so you can write vector code in assembly language or extended asms using a gnu toolchain.

CodeDance364, 2021-12-30:
thanks,.s file really works,but when I use -O -S hello.c,it seems that it only shows the default version as rv64i2p0_m2p0_a2p0_f2p0_d2p0_c2p0,but I saw it can actually support i2p1 configuration as you mentioned in issue#997,so my question is:
1.How many versions that i-extension support?where is the version list that shows all supported versions that user can choose to configure,such as 2p0,2p1.Other extensions have the same question.
2.How can we get the version number of other extensions such as B,H,since I can only see _b but without version number followed in arch info in .s file when I configured -march=rv64gcb

Thanks.

jim-wilson, 2021-12-30:
New stuff like B and H won't be supported in riscv-gnu-toolchain until after it appears in an FSF release.  Binutils will release end of January or so and GCC will release in May or so.  And then it might be a few more months after a release before we merge them into riscv-gnu-toolchain.  But if you want to use upstream top-of-tree development sources, you can see stuff earlier.  V is in upstream binutils but not upstream GCC, there is no date for that yet.  If you want V, use LLVM.

The tools don't support a lot of different versions of extensions.  Any draft versions will be dropped once we have support for the standard version.  Backward compatibility breaks, which still happen occasionally, means there is no easy way to support old versions of stuff.  So we just drop old versions which it becomes too inconvenient to continue maintaining them.

In gcc, there is a table in gcc/common/config/riscv/riscc-common.c.  E.g.
[https://gcc.gnu.org/git/?p=gcc.git;a=blob;f=gcc/common/config/riscv/riscv-common.c;h=f352ff4ce7bae0a7678d1f64ceb43932070981cc;hb=HEAD#l81](https://gcc.gnu.org/git/?p=gcc.git;a=blob;f=gcc/common/config/riscv/riscv-common.c;h=f352ff4ce7bae0a7678d1f64ceb43932070981cc;hb=HEAD#l81)
In GNU as, 
[https://sourceware.org/git/?p=binutils-gdb.git;a=blob;f=bfd/elfxx-riscv.c;h=c575ab093f962e92a1174c3ca257ca2e4279a8f5;hb=HEAD#l1140](https://sourceware.org/git/?p=binutils-gdb.git;a=blob;f=bfd/elfxx-riscv.c;h=c575ab093f962e92a1174c3ca257ca2e4279a8f5;hb=HEAD#l1140)

For most of the extensions, neither gcc nor GNU as actually do anything with the version numbers other than write them into the ELF attributes.  A simulator may care, if semantics changes from one version to the next.  But most of the changes don't affect gcc or GNU as.  One place where it does matter is priv spec versions and CSRs, as they keep changing csr names and numbers.

Note that technically, B does not exist anymore.  There are no subextensions, only extensions.  So there is only zba, zbb, etc.  Also, there seems to be some disagreement about the name for H, and whether H should refer to the old hypervisor extension or the new one, and whether the new one might perhaps need a s* name because all priv spec extensions are supposed to start with s.  Also, H (the new one I mean) might not have any actual version number associated with it.  It may take months to sort out this mess.

CodeDance364, 2022-01-15:
The links you given are exactly what I want.
Thanks very much.

closed 2022-01-15 by CodeDance364