riscv-collab/riscv-gnu-toolchain #1007: make linux ERROR

shomov, 2022-01-04:
Hello,
I'm trying to build
```
./configure --with-arch=rv32im --prefix=/etc/riscv-gnu-toolchain
make linux
```
But I got the following errors:
```
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv32-unknown-linux-gnu
checking for riscv32-unknown-linux-gnu-gcc... riscv32-unknown-linux-gnu-gcc
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether riscv32-unknown-linux-gnu-gcc  accepts -g... yes
checking for gcc... gcc
checking for riscv32-unknown-linux-gnu-readelf... riscv32-unknown-linux-gnu-readelf
checking whether we are using the GNU C++ compiler... no
checking whether this-is-not-the-compiler-youre-looking-for accepts -g... no
checking whether this-is-not-the-compiler-youre-looking-for can link programs... no
checking for sysdeps preconfigure fragments... aarch64 alpha arc arm csky hppa i386 m68k microblaze checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
mips nios2 powerpc riscv glibc requires the A extension
make: *** [Makefile:327: stamps/build-glibc-linux-rv32im-ilp32] Error 1
```

gcc version 9.3.0 (Ubuntu 9.3.0-17ubuntu1~20.04)

Thanks!

jim-wilson, 2022-01-04:
Like the message says, glibc requires the A (atomic) extension, and you tried to configure without it.  You can't have a working multicore linux system without atomic support.

shomov, 2022-01-04:
> Like the message says, glibc requires the A (atomic) extension, and you tried to configure without it. You can't have a working multicore linux system without atomic support.

I 'm sorry for the dilettante questions, but how can I solve the problem?

jim-wilson, 2022-01-04:
Instead of using --with-arch=rv32im use --with-arch=rv32ima.

closed 2022-01-04 by shomov