riscv-collab/riscv-gnu-toolchain #1008: riscv-nuclei-elf-gcc: do not fail on "-masm=att"

vazhnov, 2022-01-06:
Hello!

I'm not sure this is correct project to ask this feature.

I use workstation with GNU/Linux and x86_64 architecture.
I use vim + [vim-syntastic](https://github.com/vim-syntastic/syntastic) for syntax and style checks.

I faced with an issue: when I use `riscv-nuclei-elf-gcc` instead of system `gcc`, I receive an error message:

```
syntastic: 0.575801: checker output: [‘gcc: error: unrecognized command-line option ‘-masm=att’’, '']
```

This option, `-masm=att` came from this code https://github.com/vim-syntastic/syntastic/blob/master/syntax_checkers/asm/gcc.vim#L44 :

```vim
'main_flags': '-x assembler -fsyntax-only' . (g:syntastic_asm_generic ? '' : ' -masm=' . s:GetDialect(buf)) })
```

Is it possible to ignore this option, `-masm=att`, and not fail?

jim-wilson, 2022-01-06:
The x86 toolchain has two incompatible assembly syntaxes: att and intel.  So this is useless for RISC-V.  You should look at the sources and figure out how to change it so that it always uses the generic asm syntax for RISC-V so that it never uses the -masm option.

vazhnov, 2022-01-06:
When I created this issue, I thought RISC-V syntax uses AT&T syntax. I thought all assemblers are divided to AT&T and Intel syntax (for example, by different parameters order).

Now I see from [this table](https://en.wikipedia.org/wiki/X86_assembly_language#Syntax), RISC-V syntax is not a part of AT&T syntax: it uses the same parameters order (source before the destination), but doesn't use prefix "$" for immediate values.

@jim-wilson , Am I right?

aswaterman, 2022-01-06:
I'd just call it "RISC-V syntax".  No `$` on register specifiers; operand order is normally `dest, src`.

closed 2022-01-06 by vazhnov