riscv-collab/riscv-gnu-toolchain #1009: Creating private bare git repository of an older version

AayushMisra, 2022-01-11:
Hie,
My objective is to push an older version of the riscv-gnu-toolchain to bare repository.
What I tried:
 I checked out to an older version, removed .git directory from the main module only(as makefile expects git submodules), and pushed to the bare repository. Upon cloning from this bare repo and building the toolchain, I am unable to get all the executables, only getting riscv64-unknown-elf-gdb, riscv64-unknown-elf-gdb-add-index, riscv64-unknown-elf-run.
Can you please help me understand the issue, and guide me to achieve the objective in the most apt way.

TommyMurphyTM1234, 2022-01-11:
You probably need to show the exact sequence of commands that you executed in order to allow others to comment.

jim-wilson, 2022-01-13:
Submodules have .git dirs that point into the top level .git dir.  So removing the top level .git dir when you have submodules checked out will likely break those submodules.

You probably have to do something complicated like clone riscv-gnu-toolchain without the submodules, push it, clone and push each submodule, then modify your riscv-gnu-toolchain to use your cloned submodules.

Or simpler, just remove all of the .git dirs, and the git dir dependencies in the Makefiles, and then commit the entire thing as one tree without submodules.  But that is only useful if you won't ever want updates from upstream sources.

See also the .gitmodules file in the top level directory, which lists where the submodules come from.

AayushMisra, 2022-01-13:
I might be diverting a little maybe in the wrong direction, but I made a few observations which I think can be the reason for the problem. For the below experiments I did not manipulate the .git directories at all .
At first I cloned and built the latest sources, replacing only the Makefile with rvv-0.8 tag's Makefile. To make this work I only had to update the submodules recursively before building. Everything works fine and I get all the executables. Now coming to the issue:
1. I copy the riscv-gnu-toolchain directory to a different directory, use the latest(original with the sources) Makefile and build, I just get following executables(many missing):
                                riscv64-unknown-elf-addr2line      riscv64-unknown-elf-nm
                                riscv64-unknown-elf-ar             riscv64-unknown-elf-objcopy
                                riscv64-unknown-elf-as             riscv64-unknown-elf-objdump
                                riscv64-unknown-elf-c++filt        riscv64-unknown-elf-ranlib
                                riscv64-unknown-elf-elfedit        riscv64-unknown-elf-readelf
                                riscv64-unknown-elf-gdb            riscv64-unknown-elf-run
                                riscv64-unknown-elf-gdb-add-index  riscv64-unknown-elf-size
                                riscv64-unknown-elf-gprof          riscv64-unknown-elf-strings
                                riscv64-unknown-elf-ld             riscv64-unknown-elf-strip
                                riscv64-unknown-elf-ld.bfd
2. I copy the riscv-gnu-toolchain directory use the rvv-0.8 tag's Makefile and build, I just get following:
                                riscv64-unknown-elf-gdb            riscv64-unknown-elf-run
                                riscv64-unknown-elf-gdb-add-index
Mostly its undefined behavior, and cant get the build right. Can you explain what might be the issue and how to build with the copied sources.

jim-wilson, 2022-01-13:
Maybe you are getting a build error, and didn't notice?  Check for errors in the build log.

The Makefile is part of the source tree.  You can't use an old Makefile with a new source tree, or a new Makefile with an old source tree.  That isn't supported.

closed 2022-02-16 by AayushMisra