riscv-collab/riscv-gnu-toolchain #1011: RISCV gcc control JALR treat x5 not as link register

CodeDance364, 2022-01-15:
Hey guys,do you know whether gcc support any options to control whether x5 register as or not as link register?Thus when x5 is controlled as linked,JALR instruction will not be compiled as rs=x5,only if x5 controlled not as linked can JALR be compiled as rs=x5.

Dhrystone benchmark from riscv-test git has one code that intends only needs normal jump but not ret,but JALR instruction binary compiled by default riscv-gcc showed rd=x0,rs=x5.According to riscv-spec-20191213.pdf page22 Table2.1 downloaded from riscv official website,RAS will execute pop at this time,hardware RAS will treat this as ret.Although this will not affect the functional correctness of instruction execution,but it will definitely cause chaos of front-end RAS predicting since there is no corresponding call before here.

Anyone can make a elaboration?Thanks a lot.

jim-wilson, 2022-01-17:
You haven't shown any code, or pointed to where in riscv-test you think the code is wrong.  It doesn't make any sense to change gcc to handle function return differently.  But maybe something else is going on here?  Maybe there is an indirect jump someplace that is accidentally using x1/x5 when it would be better to not do so?  Unfortunately, I can't tell exactly what the problem is as you haven't given enough info.  The general rule is that I need enough info to reproduce in order to fix it.

It is also not clear if this is a compiler problem or an assembly source code issue.  Most of riscv-test is assembly code, and if the assembly code is wrong that isn't a gcc (or binutils) problem.  I don't think that there is any indirect jump in dhrystone C code, as indirect jump is not a standard C feature.

TommyMurphyTM1234, 2022-09-27:
Is this still a live issue for you @CodeDance364 ?
If it is then could you please provide additional info such as that mentioned by @jim-wilson please? 
Thanks.

TommyMurphyTM1234, 2022-09-29:
Since the RISC-V ISA specification (https://riscv.org/technical/specifications/) explicitly identifies `x5` as an Alternate Link Register I don't think that overriding this (potential?) use in the toolchain make sense. As such, I'm closing this issue.

closed 2022-09-29 by TommyMurphyTM1234