riscv-collab/riscv-gnu-toolchain #1014: ilp32d vs ilp32f

pphilippos, 2022-01-22:
Hello, I am a bit confused with the following part of the README:

```
ilp32d (32-bit hard-float), ilp32f (32-bit with single-precision in registers and double in memory, niche use only)
```

From, my understanding, and gcc's behaviour, ilp32f is supposed to be used with the F extension, while ilp32d requires the D extension. So, ilp32f does not sound like a niche use case, right? The above explanation, I think, contradicts the [description](https://www.sifive.com/blog/all-aboard-part-1-compiler-args) in a blog at Sifive (?).

TommyMurphyTM1234, 2022-01-22:
The description looks odd and maybe wrong to me especially when compared to the descriptions of f and d ABI options here:

https://www.sifive.com/blog/all-aboard-part-1-compiler-args

jim-wilson, 2022-01-23:
Most hardware is either no float or double float.  Very little hardware is single float only.  So in that sense, this is a niche use.

Also note that ilp32f can be used with both rv32imafc and rv32imafdc.  This is an upgrade path.  If you start with rc32imafc and then decide to upgrade the hardware to rv32imafdc but need ABI compatibility with old code, then you can continue to use ilp32f on rv32imafdc.  This is another niche use.

But anyways, this is just a readme.  I don't see the point in arguing about the wording.

pphilippos, 2022-01-23:
Thank you for the detailed explanation. Then, maybe slightly more wording is appropriate (just an opinion), in order to directly associate the ABIs with RISC-V extensions. For example, that the "niche use" refers to the extension use case (F and no D), and why ilp32f is not hard-float (?).

TommyMurphyTM1234, 2022-09-28:
I guess that this issue, and the suggested change to the text, should be logged against the SiFive blog post (https://www.sifive.com/blog/all-aboard-part-1-compiler-args) but I don't know where that should be done. I don't see the blog content represented here, for example: https://github.com/orgs/sifive/repositories. Maybe somebody from SiFive could advise?

TommyMurphyTM1234, 2022-11-10:
Closing due to lack of activity.

closed 2022-11-10 by TommyMurphyTM1234