riscv-collab/riscv-gnu-toolchain #1015: riscv64-unknown-elf-gdb: error while loading shared libraries: libpython3.8.so.1.0

i3abghany, 2022-01-22:
I downloaded the precompiled toolchain, and I ran `riscv64-unknown-elf-gdb`, and got the following error message:

```console
riscv64-unknown-elf-gdb: error while loading shared libraries: libpython3.8.so.1.0: cannot open shared object file: No such file or directory
```

The thing is, I really can't find this `libpython3.8.so.1.0` on my system. However, I have version 3.9 installed, and it's available in the shell path. I don't understand why do I need this __specific__ version of libpython, as I can't seem to find this exact file easily on the internet? I have renamed my `libpython3.9.so.1.0` to  `libpython3.8.so.1.0` and it's currently working. However, I don't think that's a sustainable solution, and it will likely silently break somewhere.

jim-wilson, 2022-01-23:
You didn't mention which precompiled toolchain that you downloaded.

The precompiled toolchain that you downloaded was compiled on a different system than what you are running, so there is a mismatch with the libpython library version.  gdb is linked against libpython, and needs the version it was linked against as sometimes there are ABI/API changes in the libpython library.  Your os might provide an older libpython as an optional package to install.  Making libpython-3.8 a symbolic link to libpython-3.9 will probably work for most cases.  python is used as a scripting language inside gdb.  Basic gdb functionality will work fine without python, but if you want to pretty-print a standard C++ type from the libstdc++ library then you need python for that.  If you are only using C, then you probably don't need python at all.

You can always try building your own toolchain from sources.

I think the SiFive toolchains come with their own libpython library to avoid this problem, but they haven't made a public release in a while, so they only have older toolchain versions available.

i3abghany, 2022-01-23:
Oops, I'm sorry for not mentioning the exact binaries that I downloaded. I added that to the issue. And yes, using the higher version has been working and I hope I don't face issues with API/ABI changes. Many thanks!

closed 2022-01-23 by i3abghany